realsenseai/librealsense #7433: realsense-viewer does not work on jetson nano

nagiton, 2020-09-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    | 4.9.140-tegra          | 
| Platform                                 |  NVIDIA Jetson nano  |
| SDK Version                            |  2.35.2   | 
| Language                            |  NA   | 
| Segment			| robot  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

realsense-viewer app get stack when i connect D435 using USB3.0 cable.

https://www.jetsonhacks.com/2019/12/22/install-realsense-camera-in-5-minutes-jetson-nano/
I followed this website and installed the SDK using installLibrealsense.sh. Just after installation, I tried to check my camera by realsense-viewer and got stack. it seems computational load getting unexpectedly high. 

when i connect camera using USB2.0 cable, this problem does not occur. 
when i connect camera to other ubuntu 18 PC, this problem also does not occur. 

if you have some idea to remedy my problem, please leave comment.
if further information needed, i will provide as far as possible.

thanks

bojankoce, 2020-09-28:
Hey, @nagiton. 
I just unboxed my Jetson Nano Devkit and installed/upgraded `librealsense` packages by following instructions from here:
https://github.com/IntelRealSense/librealsense/blob/v2.38.1/doc/distribution_linux.md

I was able to run `realsense-viewer` command with D415 camera successfully.

MartyG-RealSense, 2020-09-28:
Hi @nagiton  I have successfully used the same 5-minute install guide for the Nano that you have with my own Nano.  

When you say that you connect the camera using a USB 2.0 cable, do you mean that you are putting a USB 2 cable on the camera and connecting it to one of the Nano's four USB 3 ports please?

nagiton, 2020-09-28:
Thanks for comments. I'll give it another clean install as it may be a problem in my environment.

> Hi @nagiton I have successfully used the same 5-minute install guide for the Nano that you have with my own Nano.
>
> When you say that you connect the camera using a USB 2.0 cable, do you mean that [...]

Yes. I used USB2 cable and connect to one of nano's USB3 ports. The app indicated that it was connected to the camera via USB2.

MartyG-RealSense, 2020-09-28:
@nagiton   If the USB cable supplied with the camera works fine with your other Ubuntu 18 PC then the cable is likely okay.  It also sounds as though the camera firmware driver has no problems.

Could you check the ASIC and Projector temperatures of the camera when it is streaming in the RealSense Viewer on USB 3 please?  The temperatures are under the **Controls** sub-section of the Viewer's **Stereo Module** options.

nagiton, 2020-09-29:
@MartyG-RealSense 
I have tried the installation in another newly flashed SD and it failed. I saw the same problem.

> Could you check the ASIC and Projector temperatures of the camera when it is streaming in the RealSense Viewer on USB 3 please? The temperatures are under the Controls sub-section of the Viewer's Stereo Module options.

Sorry no, because the realsense-viewer (and also my whole device) get stuck as soon as plugging the camera in. I can not any operations when the camera connected to jetson.

By ubuntu PC, I checked ASIC and Projector temperatures, which are around 30 degrees.

Maybe I need to buy another new camera...?

MartyG-RealSense, 2020-09-29:
The camera sounds as though it is working normally on your Ubuntu PC and so does not need replacing.  The camera firmware driver should also be okay, as it is stored within the camera and so any problems with it should also appear on the Ubuntu PC.

When you use the camera with the Ubuntu PC, can you confirm please whether you are using the USB3 cable supplied with the camera or whether you are using the USB2 cable?

If you are using the USB2 cable with the Ubuntu PC, please try the USB3 cable and see whether the freeze problem with the Viewer is repeated.

nagiton, 2020-09-29:
The freeze problem does not repeat on the ubuntu PC by using both of USB2 and USB3 cable. I confirmed the Viewer shows the connection was USB3.

MartyG-RealSense, 2020-09-29:
And does the freeze still ocur if you unplug the cable from the side of the camera at its micro-size end, turn it around the other way and re-insert it, and then launch the Viewer (the micro-end supports two-way insertion).

nagiton, 2020-09-30:
I flipped the micro-size end but the problem still occur.

nagiton, 2020-10-02:
@MartyG-RealSense 
Thanks for advice.
I successfully played all of these bag files without freeze problem on my jetson nano. OpenGL seems healthy.

MartyG-RealSense, 2020-10-02:
Thanks for the information.  May I confirm then that the problem is only occurring when the camera is inserted?

nagiton, 2020-10-02:
> Thanks for the information. May I confirm then that the problem is only occurring when the camera is inserted?

Yes. The problem only occurs when
1) the camera is inserted
2) to jetson nano
3) via USB3 cable

MartyG-RealSense, 2020-10-02:
I would think that the **udev** rules that govern device handing on Linux are okay, otherwise there would likely be some form of warning message stating that there was a problem.

The Viewer freezes as soon as the camera is plugged in, before a stream is activated.  So it is not likely related to the USB port being overloaded with data. 

The firmware does not seem to be damaged as the Viewer can identify the camera on both USB2 and USB3 connections on PC.

I understand that the camera freezes as soon as it is simply plugged into the Nano on the USB3 cable, not only when the Viewer is run.  So it is not a problem with one particular program.  Is that correct, please?  

My next suggestion would be that since you have access to a PC, to try upgrading the camera's firmware to the new **5.12.8.200** firmware that was released yesterday (you should be able to do that in the Viewer on the PC as it has no problems on that machine).  As the firmware is stored inside the camera and not in the computer, it will be carried back to the Nano when you plug the camera into it.

nagiton, 2020-10-04:
> I understand that the camera freezes as soon as it is simply plugged into the Nano on the USB3 cable, not only when the Viewer is run. So it is not a problem with one particular program. Is that correct, please?

No, the problem occurs only when the viewer running.

> My next suggestion would be that since you have access to a PC, to try upgrading the camera's firmware to the new 5.12.8.200 firmware that was released yesterday (you should be able to do that in [...]

I updated the firmware to 05.12.08.200 but the problem did not resolve. After updating, the following error message appears both PC and nano right after the camera plugged in.

Invalid Value in rs2_get_option(options:0x558cb304af10, option:Emitter On Off):
hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
(option code differs between PC and nano)

I apologize that I will not work on this problem next week because of my business trip, sorry.

MartyG-RealSense, 2020-10-04:
Thanks very much @nagiton  for the additional details.  When you return from your business trip, you could consider performing a **Gold Reset** of the camera to return it to its default factory-new configuration.  This can be done with a tool called **CustomRW** that is part of the **Dynamic Calibrator** camera calibration tool for the 400 Series cameras.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

Installation instructions for Linux are provided from page 14 onwards of the user guide for the tool.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Using the **CustomRW** tool, a gold reset of the camera can be performed by launching it with the command below:

**Intel.Realsense.CustomRW -g**

nagiton, 2020-10-10:
Today, I did gold reset from my ubuntu PC. Following message shown in the terminal,

```bash
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 814412072256
  Firmware version: 05.12.08.200

Calibration on device successfully reset to default gold factory settings.
```

and pluged to the jetson nano but the problem still exists...

MartyG-RealSense, 2020-10-10:
Thank you for running the test @nagiton   It certainly is an unusual problem.  Given that after your latest firmware upgrade you encountered the same error on both Nano and PC (and the camera had been PC had been fine before the firmware update), I would next suggest downgrading the firmware back to **5.12.7.100**.  This should test whether you had a damaged firmware installed.

Also perform the update using the official short 1 meter cable supplied with the camera please, if you have not been updating firmware using that cable already.   Thanks!

MartyG-RealSense, 2020-10-20:
Hi @nagiton   Do you still require assistance with this case, please?   Thanks!

nagiton, 2020-10-20:
Thanks for many help. I will give up fixing the problem because it might be unusual.
One of these days I'll buy a new camera and try it out.

thanks!

closed 2020-10-20 by nagiton

MartyG-RealSense, 2020-10-20:
Thanks very much for the update - good luck for the future!