realsenseai/librealsense #7434: Max distance overflow bug

gwenger, 2020-09-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.0.0 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Platform                                 | PC |
| SDK Version                            |  2.36.0                         |      

### Issue Description
I am using the L515, and as best I can tell there is a bug with distances over the 'max' showing up close to the camera. I'm guessing it's something like laser returns from a previous pulse are being measured, and thus they show up close because they have traveled farther than the 'current' pulse should have. 

The included gif demonstrates what I'm trying to explain. The L515 is on a cart being pushed toward a wall. At first, the far wall shows up in roughly the middle of the hallway (based on the color of the depth map). It then gets closer and closer to the camera as the cart is pushed down the hall. Eventually the wall disappears when the sensor is at roughly its advertised max range, and then as the cart is pushed even closer, the wall reappears and snaps to the appropriate distance.

The 3D view then shows the same thing, but from the side angle it's possible to see that the 'back wall' is projected forward, floating in the middle of the hallway, and its corresponding points shrink down as the cart is moved, until they disappear and then snap to where they should be, at the appropriate size now that the projection is using the correct distance.

[image]

[image]

RealSenseSupport, 2020-10-07:
Hi @gwenger 

It seems to me what you're seeing is expected.  There may be some tuning that can be done but it doesn't seem as though it would completely remove what you're seeing.

Adding some of our tuning documentation here:
https://www.intelrealsense.com/download/13061/

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

RealSenseSupport, 2020-10-08:
Hi @gwenger 

One other adjustment that we added in the latest FW 1.5.1.3 (https://dev.intelrealsense.com/docs/firmware-releases-l500?_ga=2.198145711.1816097581.1601920876-1479303487.1578600965) and SDK 2.39.0 update was the QVGA resolution.  The optimizing the lidar camera L515 range blog was updated also to include mention of this new depth resolution.  Have a go with this updated resolution to see if the situation improves for you.

gwenger, 2020-10-21:
Didn't have access to the device for a bit, finally got a chance to play around with it today. I installed Realsense Viewer v2.39.0 and updated the firmware to 1.5.1.3. I put the device into the 'max range' preset and tested walking towards that same wall from my original gifs at the 3 possible resolutions. Basically the higher resolution, the closer that 'rollover' behavior occurs where the wall pixels snap to the correct depth.

- 1024x768 resolution -> rollover @ 7.5m
- 640x480 resolution -> rollover @ 9m
- 320x240 resolution -> here I actually hit the opposite wall (~14m) without rollover, but I found a longer hallway that exhibited the same behavior at the other resolutions and got the rollover to happen at about 15.5m

I guess I just don't understand why the distance rollover is expected behavior, and I especially don't understand why it's resolution dependent. If this is expected behavior, it seems like it limits the functionality of the device for things like mapping and navigation. I don't actually care about having terrific accuracy at these long ranges, but suddenly having an obstacle appear in front of my robot that's really just a projected image of a far away wall has caused some navigation failures, and I can't think of a way to get around that seeing as before the rollover occurs, the camera returns convincing data that there is an obstacle e.g. 1m in front of the device.

RealSenseSupport, 2020-11-13:
Hi @gwenger 

Thanks for the feedback and the detailed information.  We will keep this information in mind for the future.  The QVGA (320x240) seems like it works as expected not showing rollover past our expected 9m distance.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-12 by RealSenseSupport