realsenseai/librealsense #7435: RealSense Viewer SEH exception

tlsharkey, 2020-09-28:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```
[image]

This exception was thrown after installing the [Intel RealSense SDK for Windows](https://www.intelrealsense.com/sdk-2/) and launching the RealSense Viewer. I had just uninstalled the SDK from my computer (Intel Nuc Hades i7-8809G CPU  running Windows 10 Pro 19041.508). 2 RealSense cameras were plugged into the computer at launch. 

These cameras previously worked (viewer worked and my python scripts worked) but after leaving the computer for a month with who knows how many windows updates, both the Intel RealSense Viewer and my python scripts stopped working. They would silently fail - the scripts would fail to access the cameras, and the viewer would never start (no error messages). So I uninstalled the SDK, and installed it again from [intelrealsense.com](https://www.intelrealsense.com/sdk-2/). Upon doing so and attempting to launch the viewer, this above error occurs. Error still occurs and the application never opens. My scripts seem to be working, however - it sees 2/2 devices connected, but can only open 1/2 of the devices. This problem seems to be a recurring issue with RealSense devices however as this spotty connectivity appears on every computer I have tried RealSense devices on.

MartyG-RealSense, 2020-09-28:
Hi @tlsharkey   There was a case of this error occurring on Windows recently.  In that particular case, it was found that the Device Manager had 8 hidden instances of the 400 Series RGB and Depth driver entries.   If there was only one camera attached then there should have only been two of these drivers listed (one RGB driver and one Depth driver).  

So the additional drivers were uninstalled from the Device Manager (which can be done by right-clicking on the driver entry and selecting **Uninstall device** from the menu).

https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688837669

tlsharkey, 2020-09-29:
Doesn't seem to be the problem for me
[image]

I have 2 cameras connected and see 4 drivers (2x RGB and 2x Depth). All drivers have the same version. After uninstalling devices drivers and following instructions in [#7185 comment](https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688837669), the same error still occurs.

Other ideas?

MartyG-RealSense, 2020-09-29:
Thank you for the update.   Next, can you check that **Enable access to the camera on this device** is set to On please in the Windows privacy settings?

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

MartyG-RealSense, 2020-10-08:
Hi @tlsharkey   Do you still require assistance with this case, please?   Thanks!

tlsharkey, 2020-10-09:
Yes, still having trouble.
Camera access is enabled, and the cameras can be viewed through the windows camera app. I also have python scripts using realsense2 bindings that can see the cameras, but cannot pull frames from them through the realsense api

MartyG-RealSense, 2020-10-10:
Do you experience the same SEH error if you download and launch the standalone .exe version of the Viewer from the **Assets** list on the SDK 'Releases' page, please?

Once downloaded, this Viewer should be launched from the location that you downloaded it to, or a new Windows desktop shortcut icon created.

[image]

MartyG-RealSense, 2020-10-17:
Can you update on your current situation please @tlsharkey

MartyG-RealSense, 2020-10-25:
Do you still require assistance with this case please?   Thanks!

tlsharkey, 2020-10-25:
Never got the viewer to work, but my python scripts are working again. Don't have the time to troubleshoot further. Closing issue.

closed 2020-10-25 by tlsharkey

MartyG-RealSense, 2020-10-25:
I'm sorry to hear about the Viewer, though I am pleased to hear that you got Python functioning.

MartyG-RealSense, 2021-10-07:
Hi @sktistakis  Can you confirm whether you have checked that permission for applications to access to the camera is set to On in the Windows privacy settings please, using the method described in https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

kalai2875, 2021-11-10:
Hi I have the 'RealSense Viewer SEH exception #7435' error same as that of https://github.com/IntelRealSense/librealsense/issues/7435#issue-710568759
Attempted all the solutions mentioned there.
Still I have the same issue. Please help me to solve that issue

MartyG-RealSense, 2021-11-11:
Hi @kalai2875  Next, could you try installing the **Dynamic Calibration** tool (which includes a program called **CustomRW**) and try using CustomRW to 'gold rest' the camera to its factory defaults, as described in https://github.com/IntelRealSense/librealsense/issues/9947#issuecomment-966148418

If you have not already downloaded and installed the Dynamic Calibration tool on Windows, you can obtain it from the link below.

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html