realsenseai/librealsense #7436: Ubuntu 18 Kernel 5.4.0 Driver installation error: Failed to unload module hid_sensor_accel_3d

brendanposehn, 2020-09-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |  N/A | 
| Operating System & Version |   Ubuntu 18 | 
| Kernel Version (Linux Only)    |  5.4.0                                | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C                        | 
| Segment			|  AR                    | 

*Installation Issue*

I have been attempting to follow installation instructions according to: https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide?_ga=2.168359913.935840887.1601257305-485241723.1600379397 but when I go to apply the patched kernel modules i get the following error:

Patched kernels modules were created successfully
0
Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing uvcvideo  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
Module is resident, unloading ... modprobe: FATAL: Module hid_sensor_accel_3d is in use.
Failed to unload module hid_sensor_accel_3d. error type 1 . Operation is aborted

I have had no luck in unloading the module and am quite lost in how to remedy this problem. 

Any advice on how to install would be greatly appreciated.

ev-mp, 2020-09-29:
@brendanposehn hello, the specific error is a known issue. It will appear when during the kernel patching process RealSense devices with IMU sensor (435i,455,L515) remain connected to PC, but it can also appear appear if the devices were properly disconnected when prompted by script.
This is a kernel issue, and according to previous checks when this occurs the module `hid_sensor_accel_3d` runs in `orphan` mode, i.e. running `lsmod | grep hid_sensor_accel_3d` will show that there are no resident dependencies present.

The most effective way to rectify this is to restart Ubuntu. After boot the patched kernel modules will be already in place.
Duplicate of #5355

brendanposehn, 2020-09-29:
@ev-mp thank you very much for the quick reply. I saw that other people were solving this issue with a restart but I've had no luck doing the same. I even reinstalled completely reinstalled Ubuntu, making sure that I never plugged in my L515 at all. 

I've been referencing #5355, and have checked that `udev-rules` is installed (i.e. the file exists in my `/etc/udev/rules.d/`), but am a bit confused on where to check for `BUILD_WITH_TM2=true`.

brendanposehn, 2020-09-29:
@ev-mp I might have found a fix, let me know if you think this might cause me to run into any issues later. I added `hid_sensor_accel_3d` to my blacklisted modules with `echo -e "blacklist hid_sensor_accel_3d\n" | sudo tee -a /etc/modprobe.d/blacklist.conf `, and then rebooted and then I could run `./scripts/patch-realsense-ubuntu-lts.sh` successfully.

ev-mp, 2020-09-29:
@brendanposehn . I think blacklisting the driver is not a good idea- unless you're not interested in IMU readings.
As for the error message - it was a one-time warning.  The first time you run the script the patches modules were copied into the proper location, but just failed to be loaded immediately (w/o restart).

There is no need to rerun the script - just use the SDK regularly and check IMU readings (and revert the `hid_sensor_accel_3d`).
The next time you'll need to repeat the procedure is when you'll upgrade/downgrade kernel version.

ev-mp, 2020-12-31:
Closing as resolved

closed 2020-12-31 by ev-mp