realsenseai/librealsense #7439: Issues while building node-librealsense API

shivaniramachandra19, 2020-09-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    | NA                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.36.0.2034                        | 
| Language                            |  nodejs                     | 
| Segment			|  others                           | 

### Issue Description
According to the GitHub Instructions, I followed all the steps and pre-requisites for setting up the node JS environment.
When i execute the command: npm install --save node-librealsense or npm install node-librealsense, I frequently face issues during the last step of build. 

[image]

This is one of the many issues, but every single time the execution happens, I tend to face different issues. I would require a step by step documentation of the node-js development setup since the github instructions doesn't elaborate all the steps which is resulting into failure of the build. Even the CMake configuration is not in detail from the scratch.

I would like to use the node-librealsense-API for development of gesture controls. But unable to proceed further due to constant build failures though the complete setup is done as per the instructions. 

Any leads regarding the issue would help me in proceeding further.

Thanks,
Shivani Ramachandra

MartyG-RealSense, 2020-09-29:
Hi @shivaniramachandra19   I looked carefully at your problem, referring to the instruction document linked to below:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs

I read the instructions and wrote my own instruction manual, as I often do with complicated install procedures.  The guide below represents the work of someone who is good at writing manuals but has not actually used node.js.  So I apologize in advance for any errors.

****************

1.  Install **node.js** with the Windows 10 .msi installer file from the link below. 

https://nodejs.org/en/download/

[image]

2.  After node.js has been installed, install required modules with the commands below:

**npm install -g jsdoc     # Required for document generation
npm install -g node-gyp  # This is optional**

You will probably need to setup proxy of npm or https proxy of npm, if you don't have direct internet connection.

**Proxy of npm**
https://docs.npmjs.com/misc/config#proxy

**https proxy of npm**
https://docs.npmjs.com/misc/config#https-proxy

3.  Set up the build environment on Windows 10 with the steps in the **Setup build environment** section of the wrapper documentation in the link below:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#setup-build-environment

4.  If you are a Windows 10 user, make sure that the Visual Studio 2017 components listed in the link below are present, as the **node-gyp** module of the 'npm install' command requires them.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#build-with-cmake

5.  Build librealsense for Windows (including support for node.js) using the commands below with CMake:

cmake -DBUILD_NODEJS_BINDINGS=1 <other options...>
make  # Will build both C++ library & Node.js binding.

6.  Once the node.js wrapper has been built, skip to the **2. Run Examples** section of the instructions.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#2-run-examples

shivaniramachandra19, 2020-09-29:
@MartyG-RealSense - Thank you for the immediate response. The links provided have already been tried and I am still facing the same issue.
[image]

The requirement would be to use the node-librealsense API directly in my web project and code for air gestures. Since web project runs on node, i just need the API package to be available for further usage. But I am unable to proceed further in my research.

node command: **npm install --save node-librealsense**  would basically get the files of D415 and libraries which can be re-used for further coding is what I understood. This one step should set my libraries in place is what I understand. Correct me if i am wrong?

The node modules are not getting generating completely which is becoming a blocker for me to execute the examples provided by Intel. I have attached the screenshots of the folder structures too for better reference.

Folder structure of nodeJS wrapper:
[image]

Folder structure of node-modules under examples folder:
[image]

How this can be resolved? Should I consider any other steps? How to proceed further?

Thanks in advance ! :)
Shivani

MartyG-RealSense, 2020-09-29:
Thank you, the images were helpful.  They enabled me to locate a case that had similar log errors when **npm install** was used.  

https://github.com/IntelRealSense/librealsense/issues/1843

It was suggested to the RealSense user in that case that they try **Developer Command Prompt for VS**.  

https://github.com/IntelRealSense/librealsense/issues/1843#issuecomment-397218628

shivaniramachandra19, 2020-09-30:
@MartyG-RealSense : Hi, Thank you for all the above links. I have tried all of the above and seems like still the build throws the same error. Developer Command prompt for VS too didn't work. I am new to the realsense camera and cmake as well and unable to understand where the issue occurs. I did follow the instructions of another realsense user which you had pointed out #6270 (comment). This too didn't work on my end. Wondering what configurations are going wrong in my system. 

I have attached the screenshots again for your reference.
[image]

This is the same npm error:
[image]

Any other suggestions would be really helpful.

Thanks in advance ! :) 
Shivani

MartyG-RealSense, 2020-09-30:
I located another case about nodejs installation on Windows, where the RealSense user in that case had log errors similar to yours.  They were advised by RealSense team members that the npm package is outdated and it should work if they install from source code.  They discuss the process for doing so.

https://github.com/IntelRealSense/librealsense/issues/4242

shivaniramachandra19, 2020-10-01:
@MartyG-RealSense : Hi Marty, Thank you for the help so far. I have couple of steps mentioned in the above comments. Still I am facing an issue with the node build. I wanted to know the msbuild path configuration details if using visual studio community 2019 version. Since what i observe from the build failures, it constantly breaks when it encounters the path of msbuild. But i have verified everywhere that community 2019 version will have this configured as the system path: **C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin**

But somehow the build tends to fail at this step. I wanted to know which path was configured when #6270 (comment) realsense user was installing the nodejs build? Seems like they had used visual studio community 2019 just like in my case.

The issue always occurs when i use the **npm install** command after all the steps done as per the #6270(comment) instructions. I have even installed the latest nodejs again after referring your previous comment thinking it might be outdated.
[image]

I have attached the screenshot for your reference:
This is my admin windows powershell screen where i had done npm install of windows-build-tools too and npm install of node library.
[image]

Any hunch on this issue? How to resolve this further?

Thanks in advance ! :)
Shivani Ramachandra

MartyG-RealSense, 2020-10-02:
Apologies for the delay in responding further.   Reviewing the case again, the impression that I have is that when the advice was given that the npm package is outdated and the wrapper should be built from source code instead, the advice was meaning that the **npm install --save node-librealsense** command should not be used.

Basically, perform all of the steps in the **1.  Build From Source** section for installing the nodejs wrapper from source code and ignore the **Notice: before you go further** section at the start of the instructions.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#1-build-from-source

The principle is similar to how librealsense can be built without kernel patching using the **RSUSB** installation method over an internet connection, but the build can then break if the patch script from the manual source code installation process is run afterwards (because the RSUSB method does not need that script).

shivaniramachandra19, 2020-10-05:
@MartyG-RealSense  - Hi Marty, Apologies for the delay in response. I have already tried the official way and it didn't work for me. I am not executing npm install --save node-librealsense command. I have purely followed **Build from Source** Steps. Tried the procedure many times but the execution fails according to step: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#2-run-examples

Note: I have followed all the steps from "**Build to Source**"
The following issue occurs when i follow the official instructions mentioned in the link above: 
[image]

I followed #6270 instructions since the official ones failed to install as per your previous suggestions. When i followed #6270 instructions, i could observe the msbuild.exe path issue keeps popping up though i have set my environment variables.

No matter which procedure, npm install command remains to be common and this is where the build keeps failing. :( 

Any further clarity or insight on this would be helpful ?

Thanks in advance ! :)
Shivani Ramachandra

MartyG-RealSense, 2020-10-05:
I will seek advice within Intel about your case.  Thank you very much for your patience.

shivaniramachandra19, 2020-10-05:
Thank you so much @MartyG-RealSense for the help so far. It would be great if there can be a call support for this issue as it would actually help us to conclude the issue soon. Please let us know your convenience. 

Thanks in advance ! :)
Shivani Ramachandra

MartyG-RealSense, 2020-10-05:
Do you have a Non Disclosure Agreement (NDA) with Intel RealSense, please?

shivaniramachandra19, 2020-10-06:
@MartyG-RealSense - Hi Marty, We currently don't have any. Probably we can proceed to have one. Could you please provide the details for proceeding with the NDA?

MartyG-RealSense, 2020-10-06:
If you do not currently have an NDA then support contact is carried out solely through this GitHub forum instead of live call.

If you post a few contact details below then I can pass them on to the appropriate person for directing to those who handle NDA applications.  Such information could include:

-. Real name
-. Email address
-. Website address and company name if you have them

shivaniramachandra19, 2020-10-06:
@MartyG-RealSense  - Could we have your official email-address?

MartyG-RealSense, 2020-10-06:
Sure, let me know when you have the details and I will edit it out of my comment.

martyx.grover@intel.com

shivaniramachandra19, 2020-10-06:
@MartyG-RealSense  thank you so much for the email id. :)

MartyG-RealSense, 2020-10-06:
You are very welcome.   I will pass the details on.

shivaniramachandra19, 2020-10-06:
@MartyG-RealSense - We would be sending an email with the details. Thank you so much for the help so far Marty. :)

MartyG-RealSense, 2020-10-06:
Thank you @shivaniramachandra19 , your details have been passed on.

Regarding your current problem with building the node.js wrapper, I have filed a bug report with Intel so that the build process can be investigated officially by the RealSense team.  I understand that this will not provide an immediate solution to your problem though, so I apologise for that.

shivaniramachandra19, 2020-10-08:
Hi @MartyG-RealSense, is there any update about the NDA? and the node JS issue? 

Any insight about them would be really great ! :) 

Thanks in advance !
Shivani Ramachandra

MartyG-RealSense, 2020-10-08:
Hi @shivaniramachandra19   Your contact details have been passed to the Intel members responsible for processing RealSense NDA applications.  I do not receive information about the progress of applications.

A bug report has also been filed with Intel about the source-code build process for Node.js so that the issue can be investigated.   There is no other information that can be provided in the meantime to help you resolve the problem yourself, unfortunately.

shivaniramachandra19, 2020-10-08:
@MartyG-RealSense  - Thanks Marty for the update :) I also wanted to know whether there is any SDK of intel's which comes with inbuilt gesture recognition (most hand movement/still gestures) feature?

MartyG-RealSense, 2020-10-08:
The RealSense SDK does not have in-built gesture recognition.  Of the third-party solutions, **Nuitrack Pro** has full-body and face tracking and gesture recognition, and their new **Nuitrack AI** has enhanced next-generation tracking.

https://nuitrack.com/

https://github.com/3DiVi/nuitrack-sdk

shivaniramachandra19, 2020-10-08:
Thank you so much for the links Marty :) I would definitely take a look at them. Apart from these, I ventured your site for third-party softwares after your mention and had come across gestoos. Is it similar to nuitrack and where can we download the SDK if there is one? Just wanted to try both of them (nuitrack and gestoos-if there is any).

MartyG-RealSense, 2020-10-08:
I haven't personally used Gestoos but it is compatible with RealSense and is highlighted on Intel's website about software that supports RealSense.  It provides this link to the Gestoos website's developer page, which has a **Request the SDK:** button to request more information from Gestoos by completing a contact form:

https://gestoos.com/developer-center/?utm_source=intelrealsense_website&utm_medium=callout&utm_campaign=partner_page

shivaniramachandra19, 2020-10-13:
@MartyG-RealSense  - Hi Marty, Hope you are doing good? I just wanted to know the update about the NDA and nodejs bug. Approximately how much time it would take?

Thanks in advance ! :)
Shivani

MartyG-RealSense, 2020-10-13:
I do not receive information about the NDA registration process so I cannot provide an update about the progress of your application, unfortunately.  There are also no updates or time schedule that I can provide about Node.js after the bug report about the build failures was filed with Intel.  I do apologise.

In regard to hand gesture recognition, Nuitrack SDK is the only solution that I know of for the 400 Series cameras.  For the RealSense SR300 / SR305 there is **TouchDesigner**, which provides gesture functions in its CHOP system:

https://docs.derivative.ca/RealSense_CHOP#Parameters_-_Gestures_Page

shivaniramachandra19, 2020-10-14:
@MartyG-RealSense  - Thanks Marty. But the urge behind update is that we are totally stuck in our development and unable to proceed further. If the NDA is approved we could proceed over a call and it would be of great help to the team. Let us know your thoughts.

Thanks in advance ! :)
Shivani Ramachandra

MartyG-RealSense, 2020-10-14:
I researched your case again and found a recent case of Node.js installation problems on Windows in September 2020 that had not appeared in researches before.  In that case, the RealSense user failed to build from source in SDK 2.38.1.  

Their solution was to build with **npm install** but doing so using the commands below:

`cd <librealsense_dir>\wrappers\nodejs`
`npm i --vs-configuration=Release`

The other component of their successful solution was downgrading from Node.Js 14 to Node.js 10.22.0

https://github.com/IntelRealSense/librealsense/issues/4884

MartyG-RealSense, 2020-10-26:
Hi @shivaniramachandra19   I have an update for you.  There is a pull-request for node.js that is scheduled to be part of SDK 2.40, released about 2 weeks from the time of writing this.

https://github.com/IntelRealSense/librealsense/pull/7573

The subject of upgrading npm is also being looked at, though there is not a time schedule that can be provided for that.

shivaniramachandra19, 2020-10-28:
@MartyG-RealSense  - Thanks for the update. So is there any progress on the NDA? Apart from that, where can I take the pull request from, and is there any documentation related to the same? When will be the release of SDK 2.40 if needed to check out?

MartyG-RealSense, 2020-10-28:
@shivaniramachandra19   I do not have access to information about the progress of NDA applications.

The pull request says that the pull was yesterday merged into the **development** branch of the SDK (SDK versions are in the 'development' branch during their development and then appear on the 'master' branch when the finished version is published). You can switch branches to preview development versions of librealsense by visiting the front page of the librealsense Gitub and then left-clicking on the **Master** option to open a drop-down list of branches.

https://github.com/IntelRealSense/librealsense

[image]

SDK releases do not usually have an exact scheduled day of release.  You can be notified when a new version is published by visiting the **Releases** page and using the **Watch** option in the top corner of the page to set a notification for new releases.

https://github.com/IntelRealSense/librealsense/releases

[image]

shivaniramachandra19, 2020-11-03:
@MartyG-RealSense - Hi Marty, Sorry for the delayed response. I tried building the node wrapper with the latest development branch code. After multiple tries, seems like I was able to build to the node library. :) But there is one more issue now. When I try executing any examples except for sensor-control.js, the glfw window launches and exits within a few milliseconds. I have attached the screenshot for reference:

[image]
As seen in the screenshot, examples launche the window but exit with few codes written as part of console.logs.

It would be great if you could help me with this further.

MartyG-RealSense, 2020-11-03:
Great to hear that you made significant progress!

Is **glfw-window.js** one of the examples that does not work, please?

Does **realsense_viewer** work?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs/examples

shivaniramachandra19, 2020-11-04:
As mentioned in the screenshots of my previous comment, I am trying to execute all the examples in the examples folder. 

node nodejs-capture.js
node nodejs-align.js
node nodejs-savetodisk.js
node nodejs-pointcloud.js

All of the above did not provide any output and had just flickered a white screen with the example name and exited within milliseconds. The realsense viewer seems to work fine. But not sure why these examples wouldn't execute.

Please let me know your thoughts.

Thanks,
Shivani

MartyG-RealSense, 2020-11-04:
I will pass your crash experience with the node.js examples on to Intel to seek feedback.  Thanks again for your patience.

shivaniramachandra19, 2020-11-09:
@MartyG-RealSense  - Hi Marty, Any further update on this?

MartyG-RealSense, 2020-11-09:
@shivaniramachandra19   I am still awaiting a response from Intel.  I have just sent them a reminder.

MartyG-RealSense, 2020-11-24:
Hi @shivaniramachandra19   Intel are addressing this Node.js build issue as part of updates to Node.js that they are working on, so I will close this case now.  Thanks so much for your patience!

closed 2020-11-24 by MartyG-RealSense