realsenseai/librealsense #7442: Unable to Connect or see L515 Lidar Camera using Windows or Linux

birdistheword96, 2020-09-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | Can't Connect to Camera | 
| Operating System & Version |   Tried on Win 10 and Linux Ubuntu 18  Jetpack 4.3 | 
| Platform                                 | Tried on PC and NVIDIA Jetson XAVIER |
| SDK Version                            |   2.38.1.2223                          | 
| Language                            |  Only trying to use RealSense Viewer                       | 
| Segment			|  N/A                          | 

### Issue Description
Have only had the camera for 2 days and barely used it, but now unable to connect to the camera, or even see it within the Realsense Viewer App. It was briefly working on the Jetson, as I could see the camera and the lidar output, but when I booted it back up its morning it now refuses to communicate with any hardware. Originally tried on Jetson this morning but I know there are compatibility issues, so attempted to use the realsense-viewer on Windows 10 via the downloadable exe, but still can't even see or recognise the Camera. Widows detects that something has been plugged in but the RealSense-Viewer does not show it as an available camera.

I have not tried to install New FW and have only been using the supplied USB cable plugged into USB 3.0 ports. Have read some posts on here about troubleshooting it but nothing so far has worked.

birdistheword96, 2020-09-29:
Further to my previous comments, I followed the steps described in the [Intel Troubleshooting Guide](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#camera-not-workingdetected) but I do not see an option for 'Imaging Devices', not can I see the device listed in the device manager. 

I have tried an alternate USB_C cable as well, and tried other USB ports on the PC, to eliminate all those factors.

Has anyone else had this problem or seen it before?

ev-mp, 2020-09-29:
@birdistheword96 , the most common issue encountered with L515 is that currently it operates only in USB3 mode.
The symptoms are OS-specific and you should review the next two issues for Win and Ubuntu incorrect enumerations respectively #7215, #7224 . 
Tightening or switching the USB cable/usb port is one of the way to troubleshoot it.
Please check whether the above applies in your case and update.

birdistheword96, 2020-09-30:
Thanks @ev-mp , but unfortunately the issue does not seem to be USB3 related. I tried different USB ports and when plugged into a USB2.0 port, I can see the same errors as #7215 where the drivers are not found, but as soon as I plug it back into any of the USB3 ports then Device Manager does not show any driver errors. However, even in a USB3 port, the RealSense Software does not list it as an available device.

I tried with another USB Cable as well, just to ensure the cable hadn't been damaged, but this had no effect, the issue seems to follow the camera.

ev-mp, 2020-09-30:
@birdistheword96, please review the mentioned tickets, gather and provide logs and info for inspection:
1. Linux -
 - kernel and lsusb logs when connecting the camera
 - elaborate the SDK installation steps performed.
2. Windows - Device Manager snapshot with Camera list expanded.

birdistheword96, 2020-10-02:
hi @ev-mp , sorry for taking a while to get back to you, I briefly had the camera working on a a standard Ubunt system aswell as on the Jetson, but it has now failed again. When using linux, and issuusing the '**_dmesg -wT_**' command, I get the following errors printed out:
[Fri Oct  2 12:55:16 2020] usb 1-2.3: new low-speed USB device number 33 using tegra-xusb
[Fri Oct  2 12:55:16 2020] usb 1-2.3: Device not responding to setup address.
[Fri Oct  2 12:55:16 2020] usb 1-2.3: Device not responding to setup address.
[Fri Oct  2 12:55:16 2020] usb 1-2.3: device not accepting address 33, error -71
[Fri Oct  2 12:55:17 2020] usb 1-2.3: new low-speed USB device number 34 using tegra-xusb
[Fri Oct  2 12:55:17 2020] usb 1-2.3: Device not responding to setup address.
[Fri Oct  2 12:55:17 2020] usb 1-2.3: Device not responding to setup address.
[Fri Oct  2 12:55:17 2020] usb 1-2.3: device not accepting address 34, error -71
[Fri Oct  2 12:55:17 2020] usb 1-2-port3: attempt power cycle
[Fri Oct  2 12:55:18 2020] usb 1-2.3: new low-speed USB device number 35 using tegra-xusb
[Fri Oct  2 12:55:18 2020] usb 1-2.3: device descriptor read/64, error -32
[Fri Oct  2 12:55:18 2020] usb 1-2.3: device descriptor read/64, error -32
[Fri Oct  2 12:55:19 2020] usb 1-2-port3: Cannot enable. Maybe the USB cable is bad?
[Fri Oct  2 12:55:19 2020] usb 1-2-port3: unable to enumerate USB device

Nothing ever shows up on windows task manager, it doesn't seem to be detected at all.

To reiterate, I have tried with different USB3 ports and ONLY using USB3. These ports work fine when using a ZED2 camera, which also requires USB3 and has issues with USB2.

birdistheword96, 2020-10-02:
Have now also tried multiple USB3.1 and USB3.0 to USB-C cables, and I still get the same behaviour, it doesn't get detected.

On a few occasions, **dmesg -wT** has come up with these messages:

[Fri Oct  2 14:07:40 2020] usb 1-2.3: new high-speed USB device number 42 using tegra-xusb
[Fri Oct  2 14:07:40 2020] usb 1-2.3: New USB device found, idVendor=8086, idProduct=0b64
[Fri Oct  2 14:07:40 2020] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Oct  2 14:07:40 2020] usb 1-2.3: Product: Intel(R) RealSense(TM) 515
[Fri Oct  2 14:07:40 2020] usb 1-2.3: Manufacturer: Intel(R) RealSense(TM) Camera
[Fri Oct  2 14:07:40 2020] usb 1-2.3: SerialNumber: 00000000F0233442

However when I use realsense-viewer or rs-enumerate-devices it still thinks that no devices are connected.

birdistheword96, 2020-10-02:
Having tried Multiple PCs, with 3 different operating systems, over 5 different USB 3.0/3.1 USB-A to USB-C cables and lots of USB3.0 ports on each PC, I can't imagine that it something other than hardware. Since I have had it less than a week, I am going to RMA it and ask for a repair or a replacement.

Will update again once I have some more hardware to try on so that the issue can be confirmed and closed.

Thank you @ev-mp for your help so far.

ev-mp, 2020-10-02:
@birdistheword96 , according to the attached log device is recognized as USB2 similar to [7224](https://github.com/IntelRealSense/librealsense/issues/7224#issuecomment-683096962):
>[Fri Oct 2 14:07:40 2020] usb 1-2.3: new high-speed USB device number 42 using tegra-xusb
[Fri Oct 2 14:07:40 2020] usb 1-2.3: New USB device found, idVendor=8086, idProduct=0b64

I cannot comment on WinOS, but based on AGX log the device does communicate with host, so I would recommend to continue checking with Xavier (we can check the Windows state if you'd provide the logs later on).

Xavier debugging:
1. - The following type of errors is and indication that the USB controller is malfunctioning, and in so you'll need to reboot the board to troubleshoot it
>[Fri Oct 2 12:55:16 2020] usb 1-2.3: new low-speed USB device number 33 using tegra-xusb
[Fri Oct 2 12:55:16 2020] usb 1-2.3: Device not responding to setup address.

2. One of the hurdles with Xavier is that it comes with a single USB3 type-C port which forces to use external USB hub for peripherals. 
 - In case you're using such hub then you can try to change it as well or try one with external power.
- Connect the camera directly to the USB port and check the kernel log to see whether it is enumerated as USB2 or USB3 (You'll need Type-C <-> Type-C cable).
 - Connect non-RealSense USB3 camera and check whether it passes the enumeration stage properly
 - AGX Xavier comes with PCI-E slot so you use an extension USB3 port and connect the Camera via that port.

birdistheword96, 2020-10-06:
Hi @ev-mp , unfortunately by the time I had read that I had already packaged up and returned the device. I have now received a new one (it looks like a previous returned one but no matter), and it worked first time. After using it all morning, it is picked up on both the Jetson and the standard Ubuntu PC without any issues, so I think in this case it was likely hardware related.

Thank you for your support anyway.

closed 2020-10-06 by birdistheword96