realsenseai/librealsense #7444: Invalid Value in rs2_get_option_range and rs2_open_multiple

ZiSo89, 2020-09-29:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D400|
|**Serial Number**|ffffffffffff|
|**Firmware Version**|05.12.07.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|935723020038|
|**Firmware Update Id**|935723020038|

When I open the Viewer I got this:

Invalid Value in rs2_get_option_range(options:00000194C5A42DE0, option:Depth Units, min:-nan, max:0, step:5.66125e-43, def:-0.021266):
hwmon command 0x2c( 9 1 0 0 ) failed (response -9= No expected user action)

And if I click to toggle button I have this msg: 

Invalid Value in rs2_open_multiple(sensor:00000194C5A42DE0, profiles:00000194C9DAD828, count:1):
hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)

I did the bellow actions but nothing, 
I successfully reset the camera to factory with the command: Intel.Realsense.CustomRW.exe -g
I update the firmware successfully to  5.12
I update the SDK to 2.38.1 
I change the cable currently is usb 3.2

Hope someone will help...

MartyG-RealSense, 2020-09-29:
Hi @ZiSo89   After performing the Gold Reset, did you perform a calibration using the most recent version of the **Dynamic Calibrator** software, as suggested by the guidance in the link below about this error message?

https://www.intel.co.uk/content/www/uk/en/support/articles/000028857/emerging-technologies/intel-realsense-technology.html

ZiSo89, 2020-09-29:
> Hi @ZiSo89 After performing the Gold Reset, did you try performing a calibration using the most recent version of the **Dynamic Calibrator** software?

The calibration does not even start,

[image]

MartyG-RealSense, 2020-09-29:
The last report of the fffffff serial number problem was on a different generation of RealSense cameras back in 2016.

https://github.com/IntelRealSense/librealsense/issues/65

It is a problem that has a long history of occurring with devices other than RealSense cameras though, as googling for **serial number fffffff** shows.

Given that it is a problem not limited to RealSense, I will refer this case to Intel for further analysis.  Thanks for your patience.

ev-mp, 2020-09-29:
@ZiSo89, it is not clear which camera you're using, can you clarify on it?
As per the error message - both the queried control and the command refer to "Advanced Mode" feature - seem that in your case this is turned off.
Please make sure it is turned on in Viewer as it it is a per-requisite for changing the depth units.
[image]

ZiSo89, 2020-09-29:
If I turn off the advanced Mode:
[image]

And when I try to change the Depth units I got the error
[image]

And the camera is this one,
[image]

MartyG-RealSense, 2020-09-29:
@ZiSo89   So you can confirm that you get the same error messages and 'ffffffff' serial number whether you have Advanced Mode on or off please?

ZiSo89, 2020-09-29:
The error message pop up when I have the advance mode **on**

ev-mp, 2020-09-29:
As for the D430 module that you use - 
1. Is this the first time you activate that?
2. What are the results of running `rs-enumerate-devices -c` ?
3. I would suspect the ASIC-Module bridge is connected properly - please reconnect and verify that you hear a "click" on both sides when the 30-pin connector slides in properly

ZiSo89, 2020-09-29:
> As for the D430 module that you use -
> 
> 1. Is this the first time you activate that?
> 2. What are the results of running `rs-enumerate-devices -c` ?
> 3. I would suspect the ASIC-Module bridge is **not connected properly** - please reconnect and verify that you hear a "click" on both sides when the 30-pin connector slides in properly

Was so simple. I didn't even noticed that there is that bridge.  Thanks anyways.

closed 2020-09-29 by ZiSo89

MartyG-RealSense, 2020-09-29:
I wondered about the bridge board linkage integrity myself as it looked a little off.  Thanks so much for the help @ev-mp   :)