realsenseai/librealsense #7445: Residue after threshold filter like tail

sbilgi, 2020-09-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (5.12.6) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17) |                                        | 
| Platform                                 | PC|
| SDK Version                            |  { l2.34.1}                          | 
| Language                            |  {C++ }                          | 

### Issue Description
When I enable the threshold filter, there was residue like tail beside of moving object. What is the reason of this? 

[image]

MartyG-RealSense, 2020-09-30:
Hi @sbilgi   May I confirm please whether this is a 2D image or a 3D point cloud?

sbilgi, 2020-09-30:
Hi @MartyG-RealSense, It's disparity map 2D, also occur in depth 2D map .

MartyG-RealSense, 2020-09-30:
Having considered a range of possibilities, there is a high likelihood that the problem is caused not by the humans but by the tiled floor pattern.  RealSense 400 Series cameras can become confused when observing a repeating horizontal or vertical pattern such as a row of fence-posts (vertical) or window blinds (horizontal).  The comment at the link below, and the comments below it, suggest resources that may help to reduce the problem.

https://github.com/IntelRealSense/librealsense/issues/6713#issuecomment-651114720

dorodnic, 2020-10-04:
Is temporal-filter also enabled? This looks like expected artifact of the temporal filter

closed 2020-10-20 by MartyG-RealSense