realsenseai/librealsense #7448: Problem storing sensor and device instances in variables

Michael-Equi, 2020-09-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  0.2.0.951 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.4.0-47-generic | 
| Platform                                 | PC  |
| SDK Version                            |   2.38.1-0~realsense0.3378 500                         | 
| Language                            |  C++                        | 
| Segment			|  Robot                          | 

### Issue Description
I have recently been trying to figure out an issue I have been having that causes my software to have the following error despite rs-enumerate and realsense-viewer picking up the T265 fine:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Unable to open device interface
```

Breaking the software down into its simplest parts, I started to look for where this issue was stemming from and I noticed that the problem seemed to occur every time I stored the device or one of the sensors as a variable in my program. I think I have figured out a solution, but I am not sure why this is a problem in the first place as it seems to work fine for some examples that I have seen such as the ROS realsense node (perhaps there is some nuance that I am missing?). If someone could explain why the code acts the way it does I would be very appreciative. Thanks.

### Example 1
This code works fine and is what I ended up settling on to solve my problems.
```
rs2::context ctx;
auto list = ctx.query_devices(RS2_PRODUCT_LINE_T200);

rs2::config cfg;
auto pipe = rs2::pipeline(ctx);
cfg.enable_device(list[0].get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
auto dev = cfg.resolve(pipe).get_device();
dev.hardware_reset();

cfg.resolve(pipe).get_device().first<rs2::pose_sensor>().set_option(RS2_OPTION_ENABLE_MAPPING, true);
cfg.resolve(pipe).get_device().first<rs2::pose_sensor>().set_option(RS2_OPTION_ENABLE_POSE_JUMPING, true);

cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
pipe.start(cfg, cb);
```

### Example 2
This code breaks with the error that I stated above.
```
rs2::context ctx;
auto list = ctx.query_devices(RS2_PRODUCT_LINE_T200);

rs2::config cfg;
auto pipe = rs2::pipeline(ctx);
// Breaks with this modification
rs2::device dev = list[0];
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
dev.hardware_reset();

cfg.resolve(pipe).get_device().first<rs2::pose_sensor>().set_option(RS2_OPTION_ENABLE_MAPPING, true);
cfg.resolve(pipe).get_device().first<rs2::pose_sensor>().set_option(RS2_OPTION_ENABLE_POSE_JUMPING, true);

cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
pipe.start(cfg, cb);
```

### Example 3
This code breaks with the following error.
```
2020-09-30 02:58:52,019 WARNING [default] Could not open device Unable to open device interface
2020-09-30 02:58:52,027 WARNING [default] Could not open device Unable to open device interface
2020-09-30 02:58:52,032 WARNING [default] Could not open device Unable to open device interface
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
```
```
rs2::context ctx;
auto list = ctx.query_devices(RS2_PRODUCT_LINE_T200);

rs2::config cfg;
auto pipe = rs2::pipeline(ctx);
cfg.enable_device(list[0].get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
auto dev = cfg.resolve(pipe).get_device();
dev.hardware_reset();

// Breaks with this line
auto pose_sensor = cfg.resolve(pipe).get_device().first<rs2::pose_sensor>();
pose_sensor.set_option(RS2_OPTION_ENABLE_MAPPING, true);
pose_sensor.set_option(RS2_OPTION_ENABLE_POSE_JUMPING, true);

cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
pipe.start(cfg, cb);
```

yutaide, 2020-09-30:
@Michael-Equi 
Hello, Michael.
I'm struggling to figure out a problem that librealsense won't recognize T265 until unplug and replug T265.
Can you share your example1's full source code ?
Thank you.

yutaide, 2020-09-30:
@Michael-Equi 
I got an error regarding undifinition of 'cb'.

Michael-Equi, 2020-09-30:
@yutaide cb is just a generic callback with the signature `void cb(rs2::frame frame)` Everything else is just in main.

closed 2026-10-23 by MartyG-RealSense