realsenseai/librealsense #7449: align-advanced of C++ runs very slow

szryuu, 2020-09-30:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | L500_Series_Production_FW_1_5_0_0 | 
| Operating System & Version |  Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.38.1.2223                         | 
| Language                            |C++/Python                         | 
| Segment			| AR                         | 

### Issue Description
On same PC , align-advanced of C++ runs very slow(5fps , used Visual Studio) , but align-advanced of Python is fast(45fps , used VScode) . I don't  know why......

dorodnic, 2020-09-30:
There were couple of similar issues opened here before. Usually its fixed by building the SDK and samples in Release configuration. Let us know if it helps

RealSenseSupport, 2020-10-07:
Hi @szryuu 

Did the suggest from dorodnic help?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-10-18 by dorodnic