realsenseai/librealsense #7450: How to install intel realsense Viewer using Raspberry Pi  from source code

dingbaiyin, 2020-09-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   |  5.12.6| 
| Operating System & Version | Ubuntu   | 
| Kernel Version (Linux Only)    | 4.19.97-v7l+                                      | 
| Platform                                 | Raspberry Pi|
| SDK Version                            | 2.36.0                     | 
| Language                            |  c++                     | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to install Intel realsense viewer tools on the Raspberry Pi to view camera images ， firmware upgrades ， one-chip-self-calibration. How can I implement this function

MartyG-RealSense, 2020-09-30:
Hi @dingbaiyin   You build the Viewer from source along with other tools and examples with: 

**cmake ../ -DBUILD_EXAMPLES=true **

Details of the location of the built binary file for the Viewer are referenced in the link below:

https://github.com/IntelRealSense/librealsense/issues/1980#issuecomment-401599861

MartyG-RealSense, 2020-10-08:
Hi @dingbaiyin   Do you still require assistance with this case, please?   Thanks!

dingbaiyin, 2020-10-09:
Sorry for the late reply, I did not reply in time during the National Day holiday.The RealViewer running interface is garbled. At the same time, you can see the following errors on the terminal as follows:
[image]
[image]
pi enabled OpenGL.I don't know what caused it and how to solve it.

MartyG-RealSense, 2020-10-09:
Hi @dingbaiyin  Your problem with the corrupted fonts in the Viewer on Pi may be related to the problem in the link below:

https://github.com/IntelRealSense/librealsense/issues/7533#issuecomment-706016316

dingbaiyin, 2020-10-12:
I use Pi 3 Model B+.
I refer to this link：https://github.com/IntelRealSense/librealsense/issues/4558
First,I edited the file ~/.realsense-config.json, adding line:
"performance.font_oversample": "1",
the I run the realsense-viewer file in the "/librealsense-2.36.0/build/tools/realsense-viewer" ，but  the problem is not solved.
second, I modify the file of  /librealsense-2.36.0/common/model-views.cpp 
change  const int OVERSAMPLE = config_file::instance().get(configurations::performance::font_oversample);
to
const int OVERSAMPLE = 1;
then I make   /librealsense-2.36.0/build/tools
I re-run realsense-viewer，but the problem is not solved yet.
Is my operation right？

MartyG-RealSense, 2020-10-12:
Hi @dingbaiyin  I compared your edits to the instructions in  https://github.com/IntelRealSense/librealsense/issues/4558 and they are the same.  Another RealSense user who had the same problem with the font distortion and also tried this method reported today that the same instructions did not work for them either.  Because they were using the SDK for programming their project and the Viewer was not required for their project, they simply chose not to use the Viewer.

I mentioned the cases of the corrupted fonts on Raspberry Pi 3 Model B to Intel recently.  I will raise the subject with them again, since the instructions for the fix do not seem to be working now.

dingbaiyin, 2020-10-13:
Thank you very much for your reply. If this problem is solved on Pi 3 Model B in the future, I hope to get relevant information. I will close the issue.

closed 2020-10-13 by dingbaiyin