realsenseai/librealsense #7454: Unable to run python wrappers example read_bag_example.py with the .bag file recorded by D435.

nirojshrestha019, 2020-09-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.38.1 }                          | 
| Language                            |  python                        | 
| Segment			|                            | 

### Issue Description
I have D435 for research purpose and was reading through documentation to learn all the functionality. When I tried running the python wrapper example read_bag_example.py (https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/read_bag_example.py). I get the following error msg:

Traceback (most recent call last):
  File "test.py", line 46, in <module>
    pipeline.start(config)
RuntimeError: Couldn't resolve requests

Please note: This error only shows when I tried to run code using my own recorded .bag file using D435. The example code run fine when ran with the sample .bag files provided here: https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

MartyG-RealSense, 2020-09-30:
Hi @nirojshrestha019   The error message suggests that the program is being asked to set up a stream whose definition it does not understand (either because of typing error in the instruction or because that mode is not currently supported).

I note that a comment on line 19 of **read_bag_example.py** reads: **Remember to change the stream resolution, fps and format to match the recorded**

The default stream configuration defined in the script is a **depth** stream of 1280x720 resolution and 30 FPS speed.  What resolution and FPS speed does your own bag file use please?

nirojshrestha019, 2020-09-30:
Hi @MartyG-RealSense Marty, thank you for the quick response and the information. The .bag file was recorded in 848 x 480 resolution with the frame rate of 30 FPS. I made small change in the code line number 43 as per your suggestion:

    config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)

And it seems to be working now. The issue has been solved. Thank you so much.

closed 2020-09-30 by nirojshrestha019

MartyG-RealSense, 2020-09-30:
No problem at all.  That's great news - thanks for the update!