realsenseai/librealsense #7455: Performance problem with pyrealsense2 align.process() on Python 3.7

DanielBretzigheimer, 2020-09-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Linux Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.38.1                          | 
| Language                            |  python                          | 
| Segment			|  others                         | 

### Issue Description
I am trying to upgrade one of our projects from python 2.7 to python 3.7. While upgrading I noticed a significant decrease in performance. When examining the problem I found out, that the "align.process(frames)" method takes longer when running on python 3.7.

Example Code:
```python
import pyrealsense2 as rs
from datetime import datetime

pipeline = rs.pipeline()
pipeline.start()

while True:
    frames = pipeline.wait_for_frames()
    align = rs.align(rs.stream.color)
    align_start = datetime.utcnow()
    frames = align.process(frames)
    align_end = datetime.utcnow()

    align_elapsed = int((align_end - align_start).total_seconds() * 1000)
    print("Took: " + str(align_elapsed) + " ms")
```

When executing the above Code with python 2.7 I get 20 ms as average for the align process. For Python 3.7 the exact same code takes 100 ms.

I have already tried to reinstall pyrealsense and librealsense but it did not help. Running on Python 3.6 has the same Problem.
I hope someone can explain this to me and has a tip what to do.

Thanks in advance

MartyG-RealSense, 2020-09-30:
Hi @DanielBretzigheimer   When you built for Python 3, which installation method (distribution packages or source code) did you use to build librealsense on your Jetson, please?

If you used the distribution method to build from packages, CUDA support for accelerating align operations should be included in the packages by default.

If you built librealense from source using CMake though then you should enable CUDA support by including the instruction **- DBUILD_WITH_CUDA=true** in the CMake build statement

DanielBretzigheimer, 2020-10-01:
Hi @MartyG-RealSense, thanks for your help. That was the problem - I reinstalled with  `-DBUILD_WITH_CUDA=true` and now it's at 20 ms.

For others having the same issue: I had to make CUDA libs available, see [this](https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13#issuecomment-573976359) comment for instructions.

closed 2020-10-01 by DanielBretzigheimer

MartyG-RealSense, 2020-10-01:
Thanks so much for the update @DanielBretzigheimer and the sharing of what worked for you!