realsenseai/librealsense #7456: L515: Address Sanitizer reports null-pointer access in librealsense when starting capturing

rafaelspring, 2020-09-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |  MacOS  | 
| SDK Version                            | 2.38.1 | 
| Language                            |  C++  | 

### Issue Description
We have experienced intermittent app crashes when using the L515 (all platforms). To get to the bottom of this we compiled and ran our app with Clang's Address Sanitizer (https://clang.llvm.org/docs/AddressSanitizer.html) enabled on MacOS.

Address Sanitizer reports a null-pointer access when the camera is supposed to begin streaming (see ASAN logs below).
We have not seen this happen with any D4xx camera.

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==31754==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x000118463f7f bp 0x70000b25fe60 sp 0x70000b25fd90 T247)
==31754==The signal is caused by a READ memory access.
==31754==Hint: address points to the zero page.
    #0 0x118463f7e in std::__1::__function::__func<void dispatcher::invoke_and_wait<librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2>(librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2, std::__1::function<bool ()>, bool)::'lambda'(dispatcher::cancellable_timer), std::__1::allocator<void dispatcher::invoke_and_wait<librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2>(librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2, std::__1::function<bool ()>, bool)::'lambda'(dispatcher::cancellable_timer)>, void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer&&) hid-device.cpp:161
    #1 0x118219031 in dispatcher::dispatcher(unsigned int)::'lambda'()::operator()() const functional:1913
    #2 0x118218f3c in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, dispatcher::dispatcher(unsigned int)::'lambda'()> >(void*) type_traits:4339
    #3 0x7fff591022ea in _pthread_body (libsystem_pthread.dylib:x86_64+0x32ea)
    #4 0x7fff59105248 in _pthread_start (libsystem_pthread.dylib:x86_64+0x6248)
    #5 0x7fff5910140c in thread_start (libsystem_pthread.dylib:x86_64+0x240c)

==31754==Register values:
rax = 0x000070000b25fdd0  rbx = 0x0000000000000000  rcx = 0x0000000000000003  rdx = 0x0000000000000080  
rdi = 0x00006070001d2c58  rsi = 0x000060c00059aad8  rbp = 0x000070000b25fe60  rsp = 0x000070000b25fd90  
 r8 = 0x0000000000000080   r9 = 0x00000fffffffffff  r10 = 0x0000000000000000  r11 = 0xffffffffffffffff  
r12 = 0x0000619000407bb0  r13 = 0x00006190004079b0  r14 = 0x000060200026d510  r15 = 0x00006190004079a0  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV hid-device.cpp:161 in std::__1::__function::__func<void dispatcher::invoke_and_wait<librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2>(librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2, std::__1::function<bool ()>, bool)::'lambda'(dispatcher::cancellable_timer), std::__1::allocator<void dispatcher::invoke_and_wait<librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2>(librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2, std::__1::function<bool ()>, bool)::'lambda'(dispatcher::cancellable_timer)>, void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer&&)
Thread T247 created by T195 here:
    #0 0x11b0c902d in wrap_pthread_create (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x5402d)
    #1 0x118218e91 in std::__1::thread::thread<dispatcher::dispatcher(unsigned int)::'lambda'(), void>(dispatcher::dispatcher(unsigned int)::'lambda'()&&) __threading_support:328
    #2 0x118218c97 in dispatcher::dispatcher(unsigned int) thread:360
    #3 0x118460015 in librealsense::platform::rs_hid_device::rs_hid_device(std::__1::shared_ptr<librealsense::platform::usb_device>) concurrency.h:217
    #4 0x118462fd8 in std::__1::shared_ptr<librealsense::platform::rs_hid_device> std::__1::shared_ptr<librealsense::platform::rs_hid_device>::make_shared<std::__1::shared_ptr<librealsense::platform::usb_device>&>(std::__1::shared_ptr<librealsense::platform::usb_device>&&&) hid-device.cpp:56
    #5 0x11845fec3 in librealsense::platform::create_rshid_device(librealsense::platform::hid_device_info) memory:4706
    #6 0x11847d01d in librealsense::platform::rs_backend::create_hid_device(librealsense::platform::hid_device_info) const rsusb-backend.cpp:67
    #7 0x1181b80bb in librealsense::l500_motion::create_hid_device(std::__1::shared_ptr<librealsense::context>, std::__1::vector<librealsense::platform::hid_device_info, std::__1::allocator<librealsense::platform::hid_device_info> > const&) l500-motion.cpp:107
    #8 0x1181ba2bc in librealsense::l500_motion::l500_motion(std::__1::shared_ptr<librealsense::context>, librealsense::platform::backend_device_group const&) l500-motion.cpp:177
    #9 0x1181ca9bd in librealsense::rs515_device::rs515_device(std::__1::shared_ptr<librealsense::context>, librealsense::platform::backend_device_group const&, bool) l500-factory.cpp:45
    #10 0x1181ca579 in std::__1::shared_ptr<librealsense::rs515_device> std::__1::shared_ptr<librealsense::rs515_device>::make_shared<std::__1::shared_ptr<librealsense::context>&, librealsense::platform::backend_device_group&, bool&>(std::__1::shared_ptr<librealsense::context>&&&, librealsense::platform::backend_device_group&&&, bool&&&) memory:2153
    #11 0x1181c58df in librealsense::l500_info::create(std::__1::shared_ptr<librealsense::context>, bool) const memory:4706
    #12 0x1180cc777 in librealsense::device_info::create_device(bool) const context.h:48
    #13 0x118564a8d in rs2_create_device rs.cpp:296
    #14 0x10e95bb64 in rs2::device_list::operator[](unsigned int) const rs_device.hpp:664
    #15 0x10e953254 in rs2::device_list::device_list_iterator::operator*() const rs_device.hpp:698
    #16 0x10e94fad1 in cameraDrivers::LibrealsenseDriver::Context::refresh(rs2::device*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const::$_0::operator()() const LibrealsenseDriver.cpp:890
    #17 0x10e94e866 in decltype(std::__1::forward<cameraDrivers::LibrealsenseDriver::Context::refresh(rs2::device*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const::$_0>(fp)()) std::__1::__invoke<cameraDrivers::LibrealsenseDriver::Context::refresh(rs2::device*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const::$_0>(cameraDrivers::LibrealsenseDriver::Context::refresh(rs2::device*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const::$_0&&) type_traits:4339
    #18 0x10e94e7c2 in SensorUID std::__1::__async_func<cameraDrivers::LibrealsenseDriver::Context::refresh(rs2::device*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const::$_0>::__execute<>(std::__1::__tuple_indices<>) future:2325
    #19 0x10e94d5ae in std::__1::__async_func<cameraDrivers::LibrealsenseDriver::Context::refresh(rs2::device*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1: [... cut ...] char> > const&)::$_3> >::*)(), std::__1::__async_assoc_state<void, std::__1::__async_func<cameraDrivers::LibrealsenseDriver::connect(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)::$_3> >*> >(void*) thread:352
    #16 0x7fff591022ea in _pthread_body (libsystem_pthread.dylib:x86_64+0x32ea)
    #17 0x7fff59105248 in _pthread_start (libsystem_pthread.dylib:x86_64+0x6248)
    #18 0x7fff5910140c in thread_start (libsystem_pthread.dylib:x86_64+0x240c)
```

RealSenseSupport, 2020-10-07:
Hi @rafaelspring 

We will need to investigate this further.

itSeez, 2021-03-17:
I might be having a similar issue with the L515. When I run the `rs-hello-realsense` binary that comes with the SDK it works fine, but when I compile it myself I always get a null pointer exception. Using the debugger I see that the function `rs2_pipeline_wait_for_frames` always returns a null pointer. This behaviour is consistent on Windows.

On Linux the behaviour is a bit different. Both the precompiled and my own build of the `rs-hello-realsense` binary work; but more often than not I get the following error on the terminal:
```
xioctl (VIDIOC_S_FMT) failed Last Error: Input/output error
```

RealSenseSupport, 2021-03-22:
Thanks @itSeez for the update.

maloel, 2021-06-24:
Hi @itSeez, @rafaelspring,
We've tried reproducing with the latest version (2.47) and have failed.
Can you verify if you're still seeing the issue?
Much appreciated!

RealSenseSupport, 2021-07-19:
Issue cannot be reproduced with LRS 2.47. Closing this issue now.

closed 2021-07-19 by RealSenseSupport