realsenseai/librealsense #7457: l515 warehouse won't activate

AndreV84, 2020-09-30:
May I know if this be used in Linux environment?somehow? likely? unlikely?
It neither works in Windows 10 set up, the activation proces sstarts but never finishes;
www.intelrealsense.com/dimensional-weight-software/

RealSenseSupport, 2020-10-06:
Hi @AndreV84 

Right now the support in on Windows, no Linux support at this time.

Regarding the activation, we are looking into this.

This issue and https://github.com/IntelRealSense/librealsense/issues/7489, are they the same issue?  We can track the issue under this one or the other one.

AndreV84, 2020-10-06:
yes, thank you for followiing up!
There is only one issue with the activation, as it doesn't get through on Windows.
Is there any solution for it?

RealSenseSupport, 2020-10-06:
Hi @AndreV84 

We will need some information with regards to the camera serial number you are trying to activate as well as email address used for activation and potentially more.

In order to make sure we maintain your privacy, it would be better for this particular issue for you to submit a ticket to the following location:  https://support.intelrealsense.com/hc/en-us/requests/new

This will provide us an avenue to discuss with you those aspects without the information being on this public github ticket.

AndreV84, 2020-10-06:
the ticket won't appear to get submitted?
I submitted the form
but it said it wouldn't get submitted unless I press verification url which I did,
but it did lead to nothing but to me entering with my another email address to the forum at that I posted also previously already

AndreV84, 2020-10-06:
reference url i got after requesting creation of the new toicket was
https://support.intelrealsense.com/verification/ticket/yVTeCdO6vyxLrrX0EGF0pCcJo/

AndreV84, 2020-10-06:
(12172) case id

closed 2020-10-08 by AndreV84