realsenseai/librealsense #7459: Out of Range Value for Argument "def" - Running Open-Source Ethernet Networking Realsense Demo

kochhalm, 2020-09-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (5.12.07.100 | 
| Operating System & Version |   {Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { 2.38.1 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi, I am trying to run the Ethernet demo to allow Realsense cameras D435 available over the network. I am following this document: https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras. However, I get the following error when I connect to the camera over the network from my linux host running the realsense-viewer. 

"RealSense error calling: out of range value for argument "def" - this when I try to connect to the IP address of the eth0 interface on the Raspberry Pi. The linux Host is on 192.168.8.2 and the Rpi is on 192.168.8.102 ... both can ping each other. 

The librealsense.log on the linux host as the following info: ```30/09 16:08:03,079 ERROR [140313947535104] (types.h:313) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
 30/09 16:08:03,079 WARNING [140313947535104] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 30/09 16:08:03,084 ERROR [140313947535104] (types.h:313) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
 30/09 16:08:03,084 WARNING [140313947535104] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 30/09 16:08:03,088 ERROR [140313947535104] (types.h:313) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
 30/09 16:08:03,089 WARNING [140313947535104] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 30/09 16:08:03,090 ERROR [140313947535104] (types.h:313) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
 30/09 16:08:03,090 WARNING [140313947535104] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
200930160814.667 [librealsense]	I: Init sensor 1, option 'Backlight Compensation', value 0
200930160814.669 [librealsense]	I: Init sensor 1, option 'Brightness', value 0
200930160814.670 [librealsense]	I: Init sensor 1, option 'Contrast', value 50
200930160814.672 [librealsense]	I: Init sensor 1, option 'Exposure', value 166
200930160814.673 [librealsense]	I: Init sensor 1, option 'Gain', value 64
200930160814.675 [librealsense]	I: Init sensor 1, option 'Gamma', value 300
200930160814.676 [librealsense]	I: Init sensor 1, option 'Hue', value 0
200930160814.678 [librealsense]	I: Init sensor 1, option 'Saturation', value 64
200930160814.679 [librealsense]	I: Init sensor 1, option 'Sharpness', value 50
200930160814.680 [librealsense]	I: Init sensor 1, option 'White Balance', value 4600
200930160814.682 [librealsense]	I: Init sensor 1, option 'Enable Auto Exposure', value 1
200930160814.683 [librealsense]	I: Init sensor 1, option 'Enable Auto White Balance', value 1
200930160814.685 [librealsense]	I: Init sensor 1, option 'Frames Queue Size', value 16
200930160814.686 [librealsense]	I: Init sensor 1, option 'Power Line Frequency', value 3
200930160814.686 [librealsense]	W: null pointer passed for argument "device"
200930161615.818 [librealsense]	I: Init sensor 1, option 'Backlight Compensation', value 0
200930161615.905 [librealsense]	I: Init sensor 1, option 'Brightness', value 0
200930161615.907 [librealsense]	I: Init sensor 1, option 'Contrast', value 50
200930161615.908 [librealsense]	I: Init sensor 1, option 'Exposure', value 166
200930161615.910 [librealsense]	I: Init sensor 1, option 'Gain', value 64
200930161615.912 [librealsense]	I: Init sensor 1, option 'Gamma', value 300
200930161615.914 [librealsense]	I: Init sensor 1, option 'Hue', value 0
200930161615.916 [librealsense]	I: Init sensor 1, option 'Saturation', value 64
200930161615.918 [librealsense]	I: Init sensor 1, option 'Sharpness', value 50
200930161615.919 [librealsense]	I: Init sensor 1, option 'White Balance', value 4600
```

kochhalm, 2020-09-30:
Also, the RsServer.cpp ignores the interface address passed in the command line: `rs-server -i 192.168.8.102` ... 

```
if (arg_address.isSet()) 
        {
            ReceivingInterfaceAddr = our_inet_addr(arg_address.getValue().c_str());
        }
```
ReceivingInterfaceAddr is not used elsewhere in the code.

RealSenseCustomerSupport, 2020-10-05:
----------------------------------------------

Hi kochhalm,
When running the viewer on ubuntu system (host), you also need to install required package by following instruction
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
similar issue reported
https://github.com/IntelRealSense/librealsense/issues/7056
to allow the kernel can support Realsense camera's data format.

once all required packaged are installed, you can plug camera directly into your ubuntu PC's USB3 port to see the viewer can stream the camera properly then move to next step that access the camera remotely..

or you can running viewer on win10 PC which no need install additional package
to access the raspberry PI4 rs-server
to see if the raspberry Pi4  is also installed properly.

RealSenseSupport, 2020-10-05:
as to rs-server -i option, as you find out, there is no function implement for it yet.
you just need to launch rs-server
the client should already able to access it through viewer as long as the IP address input to the viewer is same the fixed one you configure to your raspberry

RealSenseCustomerSupport, 2020-10-19:
----------------------------------------------

Dear Kochholm,
had you been able to run the rs-server on your Raspberry Pi4?

closed 2020-10-27 by RealSenseCustomerSupport