realsenseai/librealsense #7460: SENSOR_TIMESTAMP from the metadata of depth and color frames unavailable

GergelyMarton, 2020-09-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.0 (downloaded 2020-02-26)               | 
| Language                            |  C++                        | 

### Issue Description

I could not get the SENSOR_TIMESTAMP attribute for the depth and color frames from the frames' metadata. Should this be possible with a D435? This would be useful because I need to define the depth value for an object tracked by a marker, which marker is recognized on the color frames. The depth and color frames are not in synch, the depth seems to be ahead of the color by roughly 1 frame. As far as I understand, the D435 sensors cannot be synched. This is not a problem for me as long as I have an estimate of the delay between the depth and color frames.

I'm using the following commands to retrieve depth and color:
data:rs2::frameset data = input.wait_for_frames() // input is an rs2::pipeline
auto processed = align.process(data) // align is set with RS2_STREAM_COLOR
rs2::depth_frame depth = processed.get_depth_frame()
rs2::video_frame col = processed.get_color_frame();

Checking the available attributes with the method used in the rs-save-to-disk sample code reveals that only "TIME_OF_ARRIVAL", "BACKEND_TIMESTAMP" and "ACTUAL_FPS" attributes are available for both the depth and col frames. Only these three are available for the "rs2::frame frm" frame when I run the rs-save-to-disk sample code as well.

It seems that on [dev.intelrealsense.com](https://dev.intelrealsense.com/docs) webpage the latest version of the docs about the [frame metadata](https://dev.intelrealsense.com/docs/frame-metadata#section-librealsense2-api-for-metadata) and the [code samples](https://dev.intelrealsense.com/docs/code-samples) are not consistent. The former describes an rs2_get_frame_metadata function, while the rs-save-to-disk sample uses the rs2::frame::get_frame_metadata function. Also, the rs-config-ui is not present in the samples.

Thanks.

MartyG-RealSense, 2020-10-01:
Hi @GergelyMarton   As a first step, could ou cnfirm please whether you have metadata support enabled in Windows?  The easiest way to do this is in the **RealSense Viewer** program.  If metadata is detected as being currently disabled then when the Viewer is launched, a pop-up box titled **Frame metadata disabled** should appear that asks you if you want to enable metada.  Click the **Enable** button to enable metadata support.

[image]

If metadata is currently disabled but this box does not appear when the Viewer is launched, it may be able to be triggered by unplugging the camera and plugging it into another USB port if you have one available.

Alternatively, there is also a more complex procedure for enabling metadata support on windows that pre-dates the introduction of the more convenient Viewer-based method:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

GergelyMarton, 2020-10-01:
Hi @MartyG-RealSense, thank you for your fast response. The Frame Metadata Disabled notification does not appear if I plug the device into either of my three USB ports while the RealSense Viewer is running. Does this prove with certainty that the metadata support is enabled? The Win SDK version is 10.0.18362.0. The registry looks like this:

[image]

MartyG-RealSense, 2020-10-01:
Thank very much for the information.  I note that your registry has not got the **MetadataBufferSize** entry that the documentation for adding metadata support manually says there should be (in the image below).  I will seek a second opinion from the RealSense team members who authored that metadata-enabling documentation though by tagging them into this discussion.  Thank you very much for your patience in the meantime.

@dorodnic @ev-mp

GergelyMarton, 2020-10-03:
@MartyG-RealSense, thanks, after manually adding the necessary DWORDs in the Registry, I now have access to the metadata of the sensors that I needed.

MartyG-RealSense, 2020-10-03:
Awesome news - thanks very much for the update!