realsenseai/librealsense #7461: Minimal example for using wheel odom input on T265

Michael-Equi, 2020-10-01:
### Issue Description
Does there exist a example that shows a minimal code implementation for sending wheel odometry to the T265 camera? I have been trying to figure out how to get it to work without spitting out `2020-10-01 02:04:35,519 WARNING [default] Could not open device Unable to open device interface` whenever I try to send data to the part of the camera that fuses the wheel odometry. I am doing this in C++ on Ubuntu 20.04 with librealsense version 2.38.

The only example I have found is the ROS node which I am having a very hard time parsing through and my current attempt at doing the same as what is done in the node does not seem to work.

Michael-Equi, 2020-10-07:
@doronhi Do you have anything along these lines or could provide some feedback to get it working in my application. I noticed you are the one mostly responsible for the wheel odom integration in ROS1. Thanks

doronhi, 2020-10-07:
It was a while ago and I didn't have a chance to use it. According to the error message it seems your issue is more fundamental then the wheel odometry and has more to do with the connectivity of the T265 device.
Sorry I can't be more helpful.

LinasKo, 2020-11-12:
It's an error I've been getting when, for example, I was trying to instantiate a new `rs2::context` from another thread.

I would suggest reusing objects as much as possible, e.g. by doing
```cpp
rs2::wheel_odometer od = cfg_.resolve(pipe_).get_device().first<rs2::wheel_odometer>()
```

jingshaojing, 2021-01-08:
when I want to reset T265,i have the same question. the problem happens when  I first use "pipe.stop()",then pipe.start(cfg), according to log ,T265 has only one interface, it is occupied.so,  so the interface need to be free, my solution is reference the example "rs_pose.cpp",restart t265.