realsenseai/librealsense #7462: Getting depth FOV give values far from the data sheet

tarekmuallim, 2020-10-01:
----------------------------------------------------------------------------------------------------

| Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            | 2.36.0                      | 
| Language                            |  C++                         | 
| Segment			|                         | 

### Issue Description

I am trying to get the depth FOV like that:

```
rs2::video_stream_profile depth_stream = pipeline_profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
rs2::video_stream_profile color_stream = pipeline_profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
rs2::video_stream_profile ir_stream = pipeline_profile.get_stream(RS2_STREAM_INFRARED).as<rs2::video_stream_profile>();

depth_intrinsics = depth_stream.get_intrinsics();
color_intrinsics = color_stream.get_intrinsics();
ir_intrinsics = ir_stream.get_intrinsics();

float fov[2];
rs2_fov(&depth_intrinsics, fov);
```

I have four D415 cameras. I set the resolution to 1280 x 720 .
3 of them give FOV close to 71 x 44 while only  one give 68 x 42
According to the data sheet, FOV should be 64 x 41 +-3. so I expect max FOV to be 67 but I get 71! 

are those values normal?
or there is something wrong.

MartyG-RealSense, 2020-10-01:
Hi @tarekmuallim   I used **rs-enumerrate-devices - c** to list the calibration information of my D415 and got a result of **71 x 43.72** for the depth FOV, which is almost identical to the result from three of your D415 cameras.

[image]

The FOV of RealSense cameras is sometimes quoted by official material in a two-value format instead of the three-value horizontal / vertical / diagonal description.  For example, the main RealSense camera information site quotes the D415 FOV in the **65° × 40°** format, which is consistent with the **64 x 41** values that you quoted.

[image]

The data sheet actually quotes the FOV in terms of a 3-value set (horizontal, vertical, diagonal) and gives a depth FOV of **65°x40°x72°**.  So I wonder if the diagonal value of '72' is where confusion is occurring when comparing the FOV to the 2-value format of expressing FOV

tarekmuallim, 2020-10-01:
Hi @MartyG-RealSense 
As I understand the 2 values which I get is refer to horizontal and vertical FOV.
I am right?
If that right, the values in the data sheet is different from one in the calibration information!

MartyG-RealSense, 2020-10-01:
It seems that official information from sources such as the RealSense Store and the RealSense information website quote horizontal and vertical FOV values from the data sheet's horizontal and vertical values.

[image]

[image]

Most sources (official and unofficial) quote the data sheet FOV values, meaning that those values are the most reliable guidelines for determining whether a camera's FOV is within acceptable tolerances or not.

I do not know how  **rs-enumerate-devices** calculates its two-value FOV results unfortunately.  The best advice I can give is that if the values from 3 of the cameras agree with the values from mine, then the single 4th camera is probably the one that needs attention.

MartyG-RealSense, 2020-10-08:
Hi @tarekmuallim   Do you still require assistance with this case, please?   Thanks!

tarekmuallim, 2020-10-08:
Hi @MartyG-RealSense 
No, thank you for your support.

MartyG-RealSense, 2020-10-08:
You are very wecome @tarekmuallim  - thanks for the update!

closed 2020-10-08 by MartyG-RealSense