realsenseai/librealsense #7463: Failed to update Firmware 05.11.11.100 to 05.12.7.100

Shunichi09, 2020-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.11.11.100| 
| Operating System & Version |  Ubuntu18.04 in docker  | 
| Kernel Version (Linux Only)    |   5.4.0-48-generic | 
| Platform                                 | PC |
| SDK Version                            |  LibRealSense v2.37.0  | 

### Issue Description

I tried to solve the problem about issue https://github.com/IntelRealSense/realsense-ros/issues/1408
And I tried to update my realsense firmware to latest one (05.12.7.100).

However, during the update, I got following error:

```
 01/10 17:52:58,073 ERROR [140536741566208] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 01/10 17:52:58,073 ERROR [140536741566208] (fw-update-device.cpp:128) access failed for 8086:adb uid: 2-7-12
 01/10 17:52:58,073 WARNING [140536741566208] (rs.cpp:304) null pointer passed for argument "device"
```

How do I fix this??
Are there any solutions??

MartyG-RealSense, 2020-10-01:
Hi @Shunichi09   Which method did you use when attempting to update your firmware please?  Was it the **RealSense Viewer** program, or was it the **rs-fw-update** tool?  Thank you.

Shunichi09, 2020-10-01:
@MartyG-RealSense 
Thank you for your quick reply!!

I used the Realsense-viewer program.
And, I’m extending the USB cable with non-original one. Is this problem?? In Realsense viewer   I can see the word “USB3.2”.

MartyG-RealSense, 2020-10-01:
Ideally the original 1 meter USB cable should be used for firmware updates if you still have it available.  For example, firmware update may fail if attempted with a long extension cable that contains an 'active repeater' component to boost the signal.

Shunichi09, 2020-10-01:
Thank you for your advice. I’m going to try it.

ev-mp, 2020-10-01:
@Shunichi09, the permissions error you receive could be related to the "containerized" operation mode - the permissions configuration in docker is separate from "native" Ubuntu, so it might be needed for you "inject" the udev rules into the container as well.
I think there were previous reports on github that (at least peripherally) cited similar issues, though the references I can find now  #6214 and #2402, don't provide the root-cause analysis.
You can also test it quite straight-forwardly by upgrading FW without docker.

Shunichi09, 2020-10-02:
@ev-mp Thank for your comments. As you pointed out, I figured out this error was caused by authority. I fixed my Dockerfile.
However, I got another error... even I run the same commands in host computer not the docker container. 

```
$ rs-fw-update -l
```

```
connected devices:
1) Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: 903223050309, firmware version: unknown, USB type: unknown
``` 

According to the [documents](https://dev.intelrealsense.com/docs/firmware-update-tool), I run following commands, 

```
rs-fw-update -r -f Signed_Image_UVC_5_11_6_250.bin
```

But I got this

```
RealSense error calling ():
    failed to read firmware file
```

How can I fix this...??

Shunichi09, 2020-10-02:
@MartyG-RealSense @ev-mp
Sorry My folder path is wrong...
Finally, I can update the firmware!! Thank you for your help!!

MartyG-RealSense, 2020-10-02:
Hi @Shunichi09   The term **Signed_Image_UVC_5_11_6_250.bin** refers to the name of a particular firmware file (5.11.6.250).  This command would only work if you had downloaded this exact firmware file and placed it in the same location as the **rs-fw-update** program.

Firmware files can be downloaded from the link below:

https://dev.intelrealsense.com/docs/firmware-releases

For example, if you downloaded firmware **5.12.7.100** then your firmware update command would look like this:

**rs-fw-update -r -f Signed_Image_UVC_5_12_7_100.bin**

Please note that a new firmware **5.12.8.200** was released yesterday, so you also have the option of using that, which is now the latest version.

You may also find it easier to update the firmware by launching the **RealSense Viewer** program, going to the **More** option near the top of the options side-panel, and selecting from its menu the **Update Firmware** option, providing it with the downloaded firmware file.  Or choose **Install Recommended Firmware** to automatically install the recommended firmware for that particular SDK version.

closed 2020-10-07 by Shunichi09

sivashankar28, 2023-06-27:
So I tried all of these recommendations on Github for my D415 and have had no luck. Still ISP update failed and showing up as D410

MartyG-RealSense, 2023-06-27:
Hi @sivashankar28  Please create a new case for your problem by visiting the front page of this forum at the link below and clicking the green **New Issue** button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/