realsenseai/librealsense #7467: Saving frames to bag files in python

guyfilo, 2020-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2                         | 
| Language                            |  python }                          | 

### Issue Description
As a part of an experiment I want to save different frame of the camera as bag file, in few different bag files in the same pipeline.  witch function can make that happen and how to control the output bag file name and location?

RealSenseSupport, 2020-10-09:
Hi @guyfilo ,

For a general overview you can reference the example here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

While there isn't an exactly similar Python example there should be enough corresponding functions in the wrapper to accomplish the same thing. As a starting point there is this example thought:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

Thanks

RealSenseSupport, 2020-12-23:
@guyfilo 

Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-11 by RealSenseSupport