realsenseai/librealsense #7473: Question : Is it possible to recognize a 3D object

Eleart, 2020-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1                       | 
| Language                            |  C# unity                        | 
| Segment			| AR                      | 

### Question 
Hello,

I would like to know if is it possible to recognize a 3D oject with the camera. Or if something already exists to do that ?

MartyG-RealSense, 2020-10-01:
Hi @Eleart   If you are intending to have the camera automatically detect and identify an object that it has been trained to recognize then that is certainly possible.  For example, one way to do so is with a **Deep Neural Network (DNN).**  The RealSense SDK has a few sample DNN programs based on C++ and OpenCV in its collection of example programs, like the one in the link below:

https://github.com/twMr7/rscvdnn

If you are aiming to use C# then there are fewer options available.  Here is a link to a recent discussion about use of C#:

https://github.com/IntelRealSense/librealsense/issues/7295

Please also note that if you are using a D455 and want to use Unity, there have been cases where the D455 is not being detected in the RealSense Unity wrapper at the time of writing this.  The Intel RealSense team are aware of the issue.

If the way that you had in mind for using the D455 to detect objects is different from the method described above, please describe your needs and  I will do my best to direct you to useful resources.

Eleart, 2020-10-02:
Thanks for the anwser,

I must have expressed myself badly, I would like to know if something exists to recognize a given 3D model in the space (is position , rotation compared to the camera) and after that with this informations I will able to place an Hologram over it like [Vuforia](https://www.youtube.com/watch?v=6W7_ZssUTDQ) or [VisionLib ](https://www.youtube.com/watch?v=nzC8IpFYBnc)

MartyG-RealSense, 2020-10-02:
Thanks for the clarification.  Yes, you can use a RealSense camera to detect the position and rotation of an object (its **pose**).    

The most ready-made RealSense solution for this application is the T265 Tracking Camera, which has the in-built capability to perform 6 Degrees of Freedom (6DOF) pose tracking.  As well as detecting its own pose, it can calculate the pose of an object if that object has an image tag called an **Apriltag** attached to it.  So it is not so much object recognition as tag recognition.

https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera?_ga=2.252191561.1946939599.1600596252-1665064491.1600445960#section-41b-coordinate-systems

https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-apriltag

Please bear in mind though a recent statement by Intel about the T265 that "at this time we have moved our focus to our next generation of products".  

If you would prefer a tagless approach then a RealSense 400 Series camera could be used with an AI application that is trained to identify a specific object, and its position and rotation, and work out how to handle it.  This has been used with robot arms equipped with the 400 Series camera to select specific objects and pick them up.

https://www.youtube.com/watch?v=NkcIAmvPL7Y

Eleart, 2020-10-02:
Thanks for the informations

closed 2020-10-02 by Eleart