realsenseai/librealsense #7475: export PLY point cloud (without mesh) colored by RGB image

alddiaz, 2020-10-01:
Dear support,
I've notice it's possible to export a PLY object in IRS viewer in such a way that a point cloud without mesh is exported and at the same time colored by the corresponding RGB intensities.
I want to replicate this behavior using the python "pyrealsense2", but so far the only options that seems to work are either:
a. Export PLY point cloud + mesh, colored by RGB,
b. Export PLY point cloud without mesh, colored by predefined color palettes (not equal to RGB intensities).
Just recalling that my goal would be to achieve something like:
c. Export PLY point cloud without mesh, colored by RGB.

I wonder if you could gently send me a code snippet with an example on how to achieve "c".

Thank you :)

Aldo

MartyG-RealSense, 2020-10-01:
Hi @alddiaz   There was a past case where a RealSense user created a Python script for exporting a ply with color that lacked vertex normals.

https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293

alddiaz, 2020-10-01:
> #6194 (comment)

Thank you very much for the information.
I saw the case you mentioned and it seems it was closed but somehow it
remains unsolved.
Do you know if there is a follow up case where a solution is provided?

MartyG-RealSense, 2020-10-01:
Essentially, what the commentary on the linked-to script is saying is that it saves a color ply **without** vertex normals.  Whilst that was not the intention of its creator, it seemed as though what they created accidentally may be what you are seeking to create deliberately.  Did you have something else in mind, please?

closed 2020-10-25 by MartyG-RealSense