realsenseai/librealsense #7479: Realsense viewer gives segmentation fault when run

KokXinTan, 2020-10-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    | Linux raspberrypi 5.4.51-v7+                                     | 
| Platform                                 | Raspberry Pi Compute Module 3+  |
| SDK Version                            |  2.38.1                        | 
| Language                            |  c++            | 
| Segment			|  others                           | 

### Issue Description
When starting realsense viewer, a window briefly appears, and then disappears. Then it gives a segmentation fault.
```
$ realsense-viewer
Segmentation fault
```

I followed the steps on this website 
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

MartyG-RealSense, 2020-10-02:
Hi @KokXinTan   Bearing in mind that the Pi 3b has a smaller memory capacity than the Pi 4 that the Acrobotic guide is based on, I would recommend first trying to increase the size of your Pi's swapfile to ensure that there is sufficient virtual memory for the Pi to fall back on if the real memory gets used up.

The Raspberry Pi website states that the Compute Module 3+ is based on the Raspberry Pi 3B.  

https://www.raspberrypi.org/products/compute-module-3-plus/

The Pi 3b has its own official RealSense installation guide

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

KokXinTan, 2020-10-02:
I just did a clean install and started everything from scratch so now the realsense viewer is working again. However, once I started the realsense viewer, the characters does not show up. I attached a screenshot below. 

[image]

MartyG-RealSense, 2020-10-02:
Do you mean that when you start a stream there is no image displayed in the center panel, please?

KokXinTan, 2020-10-02:
The image shows up in the center when the stream is on, so everything is working fine, except that the fonts don't show up correctly.

MartyG-RealSense, 2020-10-02:
I was able to recreate your problem in the Viewer.  It comes from having the slider of an option called **Font Samples** maximized.

Because your fonts are corrupted, I will have to describe the interface in terms of how you see it.  So I apologize in advance for strangeness!

Please open the Viewer's settings window by left-clicking on the big grey block in the top corner of the Viewer and selecting the third option down.  This should open the **Settings** window.

[image]

Left-click on the second of the tabs on top of the window to select the **Performance** options.  Then go to the topmost option within the window, which is a slider representing the **Font Samples** option.  

[image]

Hold the left mouse button down on the blue block on the slider and drag it all the way to its minimum position.

[image]

Finally, click on the middle button at the bottom of the window (representing the **Apply** button.  This will cause the Viewer's fonts to reset to normal.

[image]

KokXinTan, 2020-10-02:
I followed your instructions, but the slider for the font samples option is already at the most minimum position. The font still show up as square boxes

MartyG-RealSense, 2020-10-03:
I ran further tests.  I found that the **Font Samples** settings produce normal characters when the slider is set to 1-7 and only produce the corrupted graphics when the slider is maximized to '8'.  Thank you for eliminating the maxed slidet position as the cause of your problem.

Further research found that the fonts may be rendered corrupted like this if there is not enough memory for the **texture atlas**.  This may be corrected by making an edit to the **model-views.cpp** file and rebuilding the Viewer, or alternatively by editing the configuration file **~/.realsense-config.json** (which does not require rebuilding the Viewer).

https://github.com/IntelRealSense/librealsense/issues/4558

MartyG-RealSense, 2020-10-12:
Hi @KokXinTan   Do you still require assistance with this case, please?   Thanks!

KokXinTan, 2020-10-12:
Hi, thanks for your reply. I think I tried the method that you mentioned, which is to edit the configuration file, and it doesn't work for me. I ended up not using the viewer at since I just needed it to verify that it would run on my rpi3+. I managed to write some code that utilizes the Realsense SDK and that works fine, and I didn't bother with the viewer after that.

I'll be closing this issue now.

closed 2020-10-12 by KokXinTan

MartyG-RealSense, 2020-10-12:
Thank you very much for the update @KokXinTan  - good luck with your project!

AyanHBK123, 2020-12-20:
> Hi @KokXinTan Bearing in mind that the Pi 3 has a smaller memory capacity than the Pi 4 that the Acrobotic guide is based on, I would recommend first trying to increase the size of your Pi's swapfile to ensure that there is sufficient virtual memory for the Pi to fall back on if the real memory gets used up.
> 
> The Raspberry Pi website states that the Compute Module 3+ is based on the Raspberry Pi 3B.
> 
> https://www.raspberrypi.org/products/compute-module-3-plus/
> 
> The Pi 3b has its own official RealSense installation guide
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

I am doing installation of Reaslsense on Pi 4 4 GB and followed instruction of the following:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

but when I run the realsense-viewer command i get this error 
Segmentation fault

MartyG-RealSense, 2020-12-20:
Hi @AyanHBK123   For the Pi 4, I would recommend following the **Acrobotic** guide for Pi 4 mentioned earlier in this discussion, whose content is similar to the official Intel one.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

A pre-made librealsense SD card image for Pi 4 provided by Intel can also act as a reference when setting up your own Pi 4 installation.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

AyanHBK123, 2020-12-20:
Just did the way link said but still same error 
when i run the command 
realsense-viewer
Segmentation fault

AyanHBK123, 2020-12-20:
Just followed the steps given in the link 
firstly got this error 
creating build/temp.linux-armv7l-3.7
creating build/temp.linux-armv7l-3.7/google
creating build/temp.linux-armv7l-3.7/google/protobuf
creating build/temp.linux-armv7l-3.7/google/protobuf/pyext
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I../src -I/usr/include/python3.7m -c google/protobuf/pyext/extension_dict.cc -o build/temp.linux-armv7l-3.7/google/protobuf/pyext/extension_dict.o -Wno-write-strings -Wno-invalid-offsetof -Wno-sign-compare
google/protobuf/pyext/extension_dict.cc: In function ‘PyObject* google::protobuf::python::extension_dict::_FindExtensionByName(google::protobuf::python::ExtensionDict*, PyObject*)’:
google/protobuf/pyext/extension_dict.cc:59:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
        ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
google/protobuf/pyext/extension_dict.cc:187:7: note: in expansion of macro ‘PyString_AsStringAndSize’
   if (PyString_AsStringAndSize(arg, &name, &name_size) < 0) {
       ^~~~~~~~~~~~~~~~~~~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

Then ignored this due to comment on issue #5092

but in the end still got this error when i run command:
realsense-viewer
Segmentation fault

MartyG-RealSense, 2020-12-20:
The Acrobotic guide has worked previously for a RealSense Pi 4 user with a segmentation fault when using the Viewer.

https://github.com/IntelRealSense/librealsense/issues/7290#issuecomment-691079291

What method are you using to compile, please?  There was a case of a Pi user who had segmentation problems when compiling with **uclibc-ng** but not when compiling with **glibc**.

https://github.com/IntelRealSense/librealsense/issues/7401#issuecomment-698320089

AyanHBK123, 2020-12-23:
I refresed my SD card and following the Acrobotic guide but getting this error when i am making 
librealsense 
under command 
make -j1

the error shown is 
/home/pi/librealsense/build/third-party/libusb/libusb/core.c:23:20: fatal error: config.h: No such file or directory
compilation terminated.
make[5]: *** [CMakeFiles/usb.dir/build.make:63: CMakeFiles/usb.dir/libusb/core.c.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:73: CMakeFiles/usb.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/libusb.dir/build.make:112: libusb-prefix/src/libusb-stamp/libusb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/libusb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

please guide how to solve this

MartyG-RealSense, 2020-12-23:
Hi @KokXinTan   A solution for the **config.h** Fatal Error is suggested in the link below:

https://github.com/IntelRealSense/librealsense/issues/7822#issuecomment-731084121

AyanHBK123, 2020-12-24:
Did everything but still the same fault on this command:
realsense-viewer 
Segmentation Fault 

I flashed the SD card, re-downloaded the Raspbian Buster and followed the exact same steps but in the end got the same error. 
Any other method to solve this error ??

MartyG-RealSense, 2020-12-24:
Do you also have problems if you use the pre-made SD card image for Pi 4 instead of using the Acrobotic guide?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

AyanHBK123, 2020-12-25:
https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

I followed this command and then run the command 
realsense-viewer but still same error of:
Segmentation fault

MartyG-RealSense, 2020-12-26:
Here is another Pi 4 installation guide that you can try:

https://github.com/IntelRealSense/librealsense/issues/4950#issuecomment-538794909

AyanHBK123, 2021-01-20:
When i run this command:
python3 setup.py test --cpp_implementation

i get this error 

ImportError: libprotobuf.so.21: cannot open shared object file: No such file or directory

----------------------------------------------------------------------
Ran 41 tests in 0.014s

FAILED (errors=39)
Test failed: <unittest.runner.TextTestResult run=41 errors=39 failures=0>
error: Test failed: <unittest.runner.TextTestResult run=41 errors=39 failures=0>

rip3rs, 2021-05-27:
on RPI 4 I am able to run `realsense-viewer` with no problem. Alas when I try to use the nodejs wrapper:

```bash
cd librealsense/wrappers/nodejs/examples/

node nodejs-capture.js
```

I get segmentation fault

MartyG-RealSense, 2021-05-27:
Hi @rip3rs  After navigating to the **/wrappers/nodejs/examples/** directory, did you carry out the middle step of inputting **npm install** please?

[image]

rip3rs, 2021-05-27:
yep I did, it is working sometimes however it needs to be restarted a few times before it can work and once it did work, the window with the camera and the capture appeared but froze.

MartyG-RealSense, 2021-05-27:
@rip3rs   Which Node.js version are you using please?

SDK 2.40.0 added support for Node.js 12 but dropped support for Node.js 6.

SDK 2.44.0 added support for Node.js 14.

Also, are you using Ubuntu on your Pi (which the Node.js wrapper can offically be installed for) or Raspian OS?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/nodejs/README.md#install-nodejs