realsenseai/librealsense #7480: RuntimeError: No device connected error after install librealsense 2.38

malith1992, 2020-10-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i and D455 } | 
| Operating System & Version |   {Ubuntu 18.04 | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            |  { 2.38 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I have build librealsense 2.38 from source using the instructions given in the [https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/](url) because I wanted to use D455 camera on it. Now both the D435i and D455 cameras are not working. It give the following error message when I run this example [https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py](url)

> Traceback (most recent call last):
  File "opencv_viewer_example.py", line 19, in <module>
    pipeline.start(config)
RuntimeError: No device connected

I have run rs-enumerate-devices, but it give me this message 

>  No device detected. Is it plugged in?

MartyG-RealSense, 2020-10-02:
Hi @malith1992  It may be related to the earlier case about D455 and the Python wrapper that you reported in September 2020 regarding Windows.  Issues regarding Raspberry Pi were also mentioned by another RealSense user on that discussion.

https://github.com/IntelRealSense/librealsense/issues/7283

This led to a couple of other issues being linked:

https://github.com/IntelRealSense/librealsense/issues/7312
https://github.com/IntelRealSense/librealsense/issues/7390

I have passed your latest Jetson-related case to Intel so that they can consider it as part of their investigations  Thanks very much for the report!

malith1992, 2020-10-05:
> Hi @malith1992 It may be related to the earlier case about D455 and the Python wrapper that you reported in September 2020 regarding Windows. Issues regarding Raspberry Pi were also mentioned by [...]

The earlier case #7283 has been solved. This is a new case.  Both cameras not connected after I install Librealsense 2.38 python wrapper on Jetson Nano. Even the realsense-viewer is not detecting the devices.

MartyG-RealSense, 2020-10-05:
This may still be related to the Python wrapper if you have installed it and if the camera was being detected before the wrapper was installed.  I have seen a couple of recent cases where the camer was detected without problems until the Python wrapper was installed, and then the camera was not detected in both pyrealsense2 and in the RealSense Viewer.

malith1992, 2020-10-12:
Is there any update about the issue ?

MartyG-RealSense, 2020-10-12:
Hi @malith1992  I was awaiting a response to the comment that I posted 7 days previously and was going to do a follow-up with you today but you asked first.    :)

I mentioned your case within Intel on October 2 but do not yet have an update.  I will follow up on it.  Thanks for the reminder and for your patience.

Michael-Equi, 2020-10-17:
I am having this same issue with a Jetson nano running 18.04 and librealsense V2.39 built from source. T265 devices show up fine in the realsense viewer and rs-enumerate-devices but the D435i that I have does not get listed or show up as connected in either. When I unplug and plug back in the D435i I get the following from the dmesg logs.

```
[Sat Oct 17 12:53:21 2020] usb 2-1.1.1: USB disconnect, device number 10
[Sat Oct 17 12:53:25 2020] usb 2-1.1.1: new SuperSpeed USB device number 11 using tegra-xusb
[Sat Oct 17 12:53:25 2020] usb 2-1.1.1: New USB device found, idVendor=8086, idProduct=0b3a
[Sat Oct 17 12:53:25 2020] usb 2-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat Oct 17 12:53:25 2020] usb 2-1.1.1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[Sat Oct 17 12:53:25 2020] usb 2-1.1.1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[Sat Oct 17 12:53:25 2020] usb 2-1.1.1: SerialNumber: 944223050589
[Sat Oct 17 12:53:25 2020] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[Sat Oct 17 12:53:25 2020] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[Sat Oct 17 12:53:25 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[Sat Oct 17 12:53:25 2020] uvcvideo: Unable to create debugfs 2-11 directory.
[Sat Oct 17 12:53:25 2020] uvcvideo 2-1.1.1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[Sat Oct 17 12:53:25 2020] uvcvideo 2-1.1.1:1.0: Entity type for entity Processing 2 was not initialized!
[Sat Oct 17 12:53:25 2020] uvcvideo 2-1.1.1:1.0: Entity type for entity Camera 1 was not initialized!
[Sat Oct 17 12:53:25 2020] input: Intel(R) RealSense(TM) Depth Ca as /devices/70090000.xusb/usb2/2-1/2-1.1/2-1.1.1/2-1.1.1:1.0/input/input8
[Sat Oct 17 12:53:25 2020] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[Sat Oct 17 12:53:25 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[Sat Oct 17 12:53:25 2020] uvcvideo: Unable to create debugfs 2-11 directory.
[Sat Oct 17 12:53:25 2020] uvcvideo 2-1.1.1:1.3: Entity type for entity Processing 7 was not initialized!
[Sat Oct 17 12:53:25 2020] uvcvideo 2-1.1.1:1.3: Entity type for entity Extension 8 was not initialized!
[Sat Oct 17 12:53:25 2020] uvcvideo 2-1.1.1:1.3: Entity type for entity Camera 6 was not initialized!
[Sat Oct 17 12:53:28 2020] usb 2-1.1.1: usb_suspend_both: status 0

```

MartyG-RealSense, 2020-10-18:
Hi @Michael-Equi   I am not able to handle cases where use of T265 is involved, as another RealSense team member is responsible for those cases.  Could you re-post your question as a new case please using the **New Issue** button on the front page of the forum please, and include 'T265' in the message title?  Thanks!

**https://github.com/IntelRealSense/librealsense/issues**

Michael-Equi, 2020-10-18:
Hi @MartyG-RealSense, the issue is not with the T265 which connects fine. The camera that I am having trouble with is the D435i.

MartyG-RealSense, 2020-10-18:
Okay @Michael-Equi   If T265 is not discussed at all and the focus of the case is limited solely to D435i then I can handle your issue.  I am off-shift now and will look at your problem when I return on Monday.  Thank you very much.

Michael-Equi, 2020-10-18:
Ok, I just mentioned the T265 to provide context that might help solve the issue with the D435i but if that is not needed then yes this is entirely an issue about the D435i.

SUN-GE, 2020-10-19:
I meet the same issue which T256 works but D435 doesn't work. Have you solved this problem?

MartyG-RealSense, 2020-10-19:
@Michael-Equi  @SUN-GE   I will need some further time to explore this issue.  I appreciaye your patience!

MartyG-RealSense, 2020-10-20:
Thanks very much everyone for your patience on this case.  The conclusions that my research of past cases reached were that the best chance of success is to build librealsense from **distribution packages** instead of from source code.

Another RealSense team member suggests to first remove traces of any previous package installations that may have been performed, and then use Intel's official package installation guide for Jetson.

https://github.com/IntelRealSense/librealsense/issues/4031#issuecomment-640945697

Michael-Equi, 2020-10-20:
Hi @MartyG-RealSense, thanks for the reply. When I install Librealsense from Debian packages it works fine but it only installs version V2.35 and I have requirements that necessitate the use of V2.38 and above (partly because of https://github.com/IntelRealSense/realsense-ros/tree/eloquent). This is why I had to build from source.

MartyG-RealSense, 2020-10-20:
Hi @Michael-Equi   I checked with Intel today whether there was any method I had not heard of for installing the Python wrapper on a Jetson with package-built librealsense.  I just received confirmation that the SDK should be built from source if you want to be able to build the Python wrapper from soure, since pip is not an option on Arm.

Michael-Equi, 2020-10-21:
> Hi @Michael-Equi   I checked with Intel today whether there was any method I had not heard of for installing the Python wrapper on a Jetson with package-built librealsense.  I just received confirmation that the SDK should be built from source if you want to be able to build the Python wrapper from source, since pip is not an option on Arm.

What specific steps should I take to accomplish this? Do the instructions for building from source not also result in the python wrapper being built from source as well?

MartyG-RealSense, 2020-10-21:
When building the Python wrapper from source, you have the choice of:

1.  Building it at the same time as building librealsense from source; or

2.  Building it separately from source **after** librealsense has been built from source.

**Option 1**

When building librealsense from source with CMake, you can add the flag **-DBUILD_PYTHON_BINDINGS:bool=true** to the CMake build instruction to build the Python wrapper too.

If you have both Python 2.7 and a version of Python 3 (such as Python 3.7) installed on your computer then you should also add a flag called **-DPYTHON_EXECUTABLE=/usr/bin/python3** to tell CMake which Python version you will be using pyrealsense2 with. 

A CMake build statement to build librealsense and the Python wrapper together from source with the RSUSB installation method (no patching required) may therefore look something like the one below.  It builds librealsense with the Python wrapper, the example programs, with CUDA support and as a Release type build that is more optimized in its running.

**cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true**

To empasize again, you do not need to kernel-patch librealsense when using DFORCE_RSUSB_BACKEND=ON

**Option 2**

If you wish to build the Python wrapper from source separately after librealsense has already been built from source, you should use the guide at the link below:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

Michael-Equi, 2020-10-22:
That worked, thank you!

MartyG-RealSense, 2020-10-22:
Great news, thanks for the update!  Is it okay to close this case now if you are satisifed with the outcome?

Michael-Equi, 2020-10-22:
As long as that solves the problem for @SUN-GE and @malith1992.

MartyG-RealSense, 2020-10-22:
Ah yes, the case was originated by @malith1992 - thanks for pointing that out!

@malith1992  and @SUN-GE   Does the advice posted earlier help to resolve your own RuntimeError problem, please?

malith1992, 2020-10-23:
I have followed the steps and it works perfectly. Thank you for taking care of this issue. @MartyG-RealSense

closed 2020-10-23 by malith1992

MartyG-RealSense, 2020-10-23:
Great to hear @malith1992 - thanks very much for the update!

SUN-GE, 2020-10-23:
I have used realsense-ros to make it work. And I will try to follow the step as you suggested. Thanks for your help.

MartyG-RealSense, 2020-10-23:
Thanks @SUN-GE  !