realsenseai/librealsense #7481: L515 crash on `pipe.stop()` / `~rs2::context`

Faaux, 2020-10-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version |   Win 10  |                                 | 
| Platform                                 | PC  |
| SDK Version                            |  2.39.0, 2.38.1 | 
| Language                            |  C++    | 

### Issue Description
Hi everyone,
I just tried to run a minimal example application to check out the L515. I immediately ran into a crash running:

```c++
rs2::context ctx;
rs2::device_hub hub( ctx );
auto dev = hub.wait_for_device();
rs2::pipeline pipe;
pipe.start();
pipe.stop();
```

Destruction of the rs2::context throws. Removing  `pipe.stop();` results in the same behavior. The same compiled code works fine with a D435.

Starting and stopping the depth sensor in the Viewer works fine. I attached a log output below.

[RS2Log.txt](https://github.com/IntelRealSense/librealsense/files/5317101/RS2Log.txt)

ev-mp, 2020-10-04:
@Faaux , that sounds like a bug.
We'll take it for internal review and update.

Did you try to put a sleep between start and stop ? I think that adding 1-3 sec may work here.
If it does then you can use it as a work-around.

Nir-Az, 2020-10-04:
@Faaux  
We tried to reproduce it with the same configuration you mentioned and we could not.
Can you share more information?

* Is the device connected via USB2 or USB3? (port and cable, can be found next to the device name in the Viewer application) 
* What is the exception details?

Faaux, 2020-10-05:
Hey @Nir-Az and @ev-mp, thanks for your quick replies.

The L515 is attached with USB 3.2 with the included cable.

After reading that you cannot reproduce I investigated a little more and came up with the following reproduction (which apparently seems to be linked to SDL2? Or a mixture of both?). Anyways on to the details:

```c++
#define SDL_MAIN_HANDLED
#include <SDL.h>

#pragma warning(push)
#pragma warning(disable : 4996)
#include <librealsense2/rs.hpp>
#pragma warning(pop)

#include <iostream>

int main() {
  try {

    rs2::log_to_file(rs2_log_severity::RS2_LOG_SEVERITY_ALL,
                     "librealsense.log");

    if (SDL_Init(SDL_INIT_VIDEO) < 0) {
      return 1;
    }

    auto window =
        SDL_CreateWindow("SDL-Realsense-Crash", SDL_WINDOWPOS_CENTERED,
                         SDL_WINDOWPOS_CENTERED, 1280, 720, 0);
    if (!window) {
      return 2;
    }

    rs2::context ctx;
    rs2::device_hub hub(ctx);
    auto dev = hub.wait_for_device();
    rs2::pipeline pipe;
    pipe.start();
    pipe.stop();
    return 0;

  } catch (rs2::error e) {
    // I am never hit
    std::cout << e.what() << std::endl;
  } catch (...) {
    // I am never hit
    std::cout << "Unknown exception" << std::endl;
  }
  return 3;
}
```

The snippet will crash the application with an `Access violation` (See the callstack at the end of the comment). This seems to be related to the creation of the SDL Window `SDL_CreateWindow(...)`, without the window creation the software shows the expected behavior. I am a very baffled by this. 

Looking through the output window I can see quite a few exceptions being thrown (but apparently being handled):
```
(KernelBase.dll) in RealsenseRepro.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
```
and
```
RealsenseRepro.exe: Microsoft C++ exception: librealsense::invalid_value_exception at memory location 0x000000CD7C71D620.
```

[librealsense.log](https://github.com/IntelRealSense/librealsense/files/5326085/librealsense.log)

Callstack:
``` 
 	realsense2.dll!00007ffa879ce834()	Unknown
 	realsense2.dll!00007ffa878cc962()	Unknown
 	rpcrt4.dll!00007ffaf58c7883()	Unknown
 	rpcrt4.dll!00007ffaf592b4a6()	Unknown
 	rpcrt4.dll!00007ffaf58a5419()	Unknown
 	combase.dll!00007ffaf494e9ff()	Unknown
 	rpcrt4.dll!00007ffaf58a97cb()	Unknown
 	combase.dll!00007ffaf48bc8e3()	Unknown
 	combase.dll!00007ffaf48bc6d4()	Unknown
 	combase.dll!00007ffaf4951646()	Unknown
 	combase.dll!00007ffaf48ec8c5()	Unknown
 	combase.dll!00007ffaf493d622()	Unknown
 	combase.dll!00007ffaf4927150()	Unknown
 	combase.dll!00007ffaf4926a31()	Unknown
 	combase.dll!00007ffaf490de01()	Unknown
 	user32.dll!00007ffaf4be7536()	Unknown
 	user32.dll!00007ffaf4be7052()	Unknown
 	combase.dll!00007ffaf48c9fc5()	Unknown
 	combase.dll!00007ffaf48cb4b6()	Unknown
 	combase.dll!00007ffaf48c8de1()	Unknown
 	combase.dll!00007ffaf48c996d()	Unknown
 	combase.dll!00007ffaf48c928e()	Unknown
 	combase.dll!00007ffaf493c636()	Unknown
 	RTWorkQ.dll!00007ffad6b47341()	Unknown
 	RTWorkQ.dll!00007ffad6b46d3c()	Unknown
 	RTWorkQ.dll!00007ffad6b46c99()	Unknown
 	realsense2.dll!00007ffa878d0e05()	Unknown
 	realsense2.dll!00007ffa878d28a4()	Unknown
 	realsense2.dll!00007ffa8794dc5d()	Unknown
 	realsense2.dll!00007ffa8798f64c()	Unknown
 	realsense2.dll!00007ffa879a473b()	Unknown
>	RealsenseRepro.exe!std::_Ref_count_resource<rs2_context *,void (__cdecl*)(rs2_context *)>::_Destroy() Line 600	C++
 	RealsenseRepro.exe!std::_Ref_count_base::_Decref() Line 542	C++
 	RealsenseRepro.exe!std::_Ptr_base<rs2_context>::_Decref() Line 774	C++
 	RealsenseRepro.exe!std::shared_ptr<rs2_context>::~shared_ptr<rs2_context>() Line 1033	C++
 	RealsenseRepro.exe!rs2::context::~context()	C++
 	RealsenseRepro.exe!main() Line 29	C++

```

Nir-Az, 2020-11-24:
@Faaux  The issue was fixed and can not be reproduced on LRS released version [2.40.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0)
Please confirm the fix on your side.

Thanks.

RealSenseSupport, 2021-01-12:
Hi @Faaux 

Were you able to confirm?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-19 by RealSenseSupport