realsenseai/librealsense #7482: SR300 recorded bag file

lghasemzadeh, 2020-10-02:
Hello,
I have recorded both RGB and IR streams at the same time into a bag file.
Now I want to watch the recorded videos, but not by the realsense viewer, I also don't want to do screen recording. I mean I want to watch them like mp4 or any other formats like that.
I checked other similar questions but I didn't get my answer.

MartyG-RealSense, 2020-10-02:
Hi @lghasemzadeh  It is certainly possible to write your own application to play back bag files.  An example of that is the Python sample program **read_bag_example.py** and the **rosbag reader** program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

https://github.com/IntelRealSense/librealsense/issues/2215

The RealSense Viewer provides the best multimedia experience for bag playback though.  As the Viewer is open-source code, perhaps you could create a custom Viewer and modify the interface to suit your exact requirements.

Alternatively, you could record the bag in ROS and then use the **make_video.py** tool of the **bag_tools** tool suite to convert the bag to video.

http://wiki.ros.org/bag_tools?distro=hydro#make_video.py

lghasemzadeh, 2020-10-06:
I tried the first link but it didn't work. So I installed ROS and Rviz, and I am able to watch the bag videos now.

MartyG-RealSense, 2020-10-06:
Glad you found a solution - thanks very much for the update!

MartyG-RealSense, 2020-10-14:
Closed due to solution found and no further comments received.

closed 2020-10-14 by MartyG-RealSense