realsenseai/librealsense #7483: Using the left camera of the stereo pair for RGB

logidelic, 2020-10-02:
In https://www.intelrealsense.com/wp-content/uploads/2019/11/BKMs_Tuning_RealSense_D4xx_Cam.pdf it says:

> While the D415 and D435 camera both have a built-in 3rd color camera that is processed through an ISP to give good color independent of the depth, there are many reasons to use the left camera of the stereo pair instead (or in addition).

However, it's not clear how to do so. Is this supported by the library?

Thanks!

MartyG-RealSense, 2020-10-02:
Hi @logidelic   You can use the RGB channel of the left camera by defining an infrared stream but using the format **RGB8** instead of Y8.  Some example C++ code for setting up such a stream is below:

cfg.enable_stream(RS2_STREAM_INFRARED, 0, width, height, RS2_FORMAT_RGB8, 30);

dorodnic, 2020-10-06:
RGB from infrared is only supported for D415 (and D455 with latest firmware). For D435(i) left camera is referring to Y8 infrared format

closed 2020-10-12 by logidelic