realsenseai/librealsense #7484: null pointer dereference in Android

joeyfedor, 2020-10-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version | Android 9 | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | Android  |
| SDK Version                           |  2.39.0 | 
| Language                               | Java | 
| Segment			         |  Drone | 

### Issue Description

When trying to running the application I get the following tombstone it occurs around 30 seconds into running the camera and can very from 5 seconds to around 3 minutes. This happens on several different phones, Android 9 or 10.
---
### Build.Gradle:
        applicationId "com.example.optiflow"
        minSdkVersion 24
        targetSdkVersion 27
----
### Important code segments:
```
            try (FrameReleaser fr = new FrameReleaser()) {
                FrameSet frames = mPipeline.waitForFrames().releaseWith(fr);
                FrameSet orgSet = frames.releaseWith(fr).applyFilter(mPointCloud).releaseWith(fr);
                FrameSet processedSet = frames.releaseWith(fr)
                        .applyFilter(mDecimationFilter).releaseWith(fr)
                        .applyFilter(mDisparity).releaseWith(fr)
                        .applyFilter(mSpatialFilter).releaseWith(fr)
                        .applyFilter(mTemporalFilter).releaseWith(fr)
                        .applyFilter(mHoleFillingFilter).releaseWith(fr)
                        .applyFilter(mDisparity).releaseWith(fr)
                        .applyFilter(mColorizer).releaseWith(fr);

                //Frame processed = processedSet.first(StreamType.DEPTH,StreamFormat.XYZ32F).releaseWith(fr);

                Frame colorDepthFrame = processedSet.first(StreamType.DEPTH).releaseWith(fr);
                Frame texture = processedSet.first(StreamType.COLOR, StreamFormat.RGB8).releaseWith(fr);
                DepthFrame depth = orgSet.first(StreamType.DEPTH, StreamFormat.Z16).releaseWith(fr).as(Extension.DEPTH_FRAME);
                Frame pointCloudFrame = orgSet.first(StreamType.DEPTH, StreamFormat.XYZ32F).releaseWith(fr);
```
---

```
        mRsContext = new RsContext();
        mRsContext.setDevicesChangedCallback(mListener);

        mPipeline = new Pipeline();

        //filters
        mAlign = new Align(StreamType.COLOR);
        mColorizer = new Colorizer();
        mDecimationFilter = new DecimationFilter();
        mHoleFillingFilter = new HoleFillingFilter();
        mDisparity = new DisparityTransformFilter(true);
        mThresholdFilter = new ThresholdFilter();
        mTemporalFilter = new TemporalFilter();
        mSpatialFilter = new SpatialFilter();
        mPointCloud = new Pointcloud();

        //config filters
        mThresholdFilter.setValue(Option.MIN_DISTANCE, 0.1f);
        mThresholdFilter.setValue(Option.MAX_DISTANCE, 7.0f);

        mDecimationFilter.setValue(Option.FILTER_MAGNITUDE, 1f);

        try (DeviceList dl = mRsContext.queryDevices()) {
            if (dl.getDeviceCount() > 0) {
[... 63 lines cut ...]
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #32 pc 0000000000145fec  /apex/com.android.runtime/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+244) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #33 pc 00000000002e2d7c  /apex/com.android.runtime/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+384) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #34 pc 00000000002ddfdc  /apex/com.android.runtime/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+892) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #35 pc 00000000005a0944  /apex/com.android.runtime/lib64/libart.so (MterpInvokeVirtual+648) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #36 pc 0000000000131814  /apex/com.android.runtime/lib64/libart.so (mterp_op_invoke_virtual+20) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #37 pc 0000000000500e52  /system/framework/framework.jar (com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run+22)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #38 pc 00000000002b4090  /apex/com.android.runtime/lib64/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEbb.llvm.6428152582483226987+240) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #39 pc 0000000000591eac  /apex/com.android.runtime/lib64/libart.so (artQuickToInterpreterBridge+1032) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #40 pc 0000000000140468  /apex/com.android.runtime/lib64/libart.so (art_quick_to_interpreter_bridge+88) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #41 pc 0000000000abb884  /system/framework/arm64/boot-framework.oat (com.android.internal.os.ZygoteInit.main+3012) (BuildId: b83295acd0cffe0b444f4a83e28d674ed6ad9261)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #42 pc 00000000001375b8  /apex/com.android.runtime/lib64/libart.so (art_quick_invoke_static_stub+568) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #43 pc 000000000014600c  /apex/com.android.runtime/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+276) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #44 pc 00000000004b04c8  /apex/com.android.runtime/lib64/libart.so (art::(anonymous namespace)::InvokeWithArgArray(art::ScopedObjectAccessAlreadyRunnable const&, art::ArtMethod*, art::(anonymous namespace)::ArgArray*, art::JValue*, char const*)+104) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #45 pc 00000000004b00b8  /apex/com.android.runtime/lib64/libart.so (art::InvokeWithVarArgs(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, std::__va_list)+408) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #46 pc 00000000003babc0  /apex/com.android.runtime/lib64/libart.so (art::JNI::CallStaticVoidMethodV(_JNIEnv*, _jclass*, _jmethodID*, std::__va_list)+624) (BuildId: 90da9848a931e540dc5b6e20456d8172)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #47 pc 00000000000ee05c  /system/lib64/libandroid_runtime.so (_JNIEnv::CallStaticVoidMethod(_jclass*, _jmethodID*, ...)+116) (BuildId: bf2a2e60ab44f8b87a30ea9dbfa5a2c6)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #48 pc 00000000000f117c  /system/lib64/libandroid_runtime.so (android::AndroidRuntime::start(char const*, android::Vector<android::String8> const&, bool)+792) (BuildId: bf2a2e60ab44f8b87a30ea9dbfa5a2c6)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #49 pc 0000000000003510  /system/bin/app_process64 (main+1216) (BuildId: 0ff9c3164ec1880cd2c0c2eee9729fcf)
2020-10-02 11:21:09.183 10139-10139/? A/DEBUG:       #50 pc 000000000007e898  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+108) (BuildId: 27147c78d1670fec7bce21c48e84eb70)
2020-10-02 11:21:09.379 1110-1110/? E//system/bin/tombstoned: Tombstone written to: /data/tombstones/tombstone_08
2020-10-02 11:21:09.401 754-754/? E/audit: type=1701 audit(1601655669.396:649): auid=4294967295 uid=10519 gid=10519 ses=4294967295 subj=u:r:untrusted_app_27:s0:c512,c768 pid=9776 comm="xample.optiflow" exe="/system/bin/app_process64" sig=11 res=1
2020-10-02 11:21:09.481 1249-1785/? E/InputDispatcher: channel '45cf5d0 com.example.optiflow/com.example.optiflow.LiveGraphing (server)' ~ Channel is unrecoverably broken and will be disposed!
2020-10-02 11:21:09.520 1249-1785/? E/InputDispatcher: channel '8e498b6 com.example.optiflow/com.example.optiflow.NavPage (server)' ~ Channel is unrecoverably broken and will be disposed!
```
---

MartyG-RealSense, 2020-10-03:
Hi @joeyfedor   This seems similar to your tombstone case from 29 July 2020, with the problem occurring soon after launch and the same block of post-processing filter code quoted.  In that case, I suggested removing the long list of post-processing filters (or at least keeping the **Decimation** filter) and seeing whether that affects stability positively.

https://github.com/IntelRealSense/librealsense/issues/6972

Do you believe that this case is a different situation to the case in July please?

joeyfedor, 2020-10-03:
Hi Marty,

Looking at the two further it does seem like the same or similar issue. I removed all the filters other than colorize and decimation and I am getting the same behavior. It does not extend the length of runtime either so at first glance they seem unrelated to me.

MartyG-RealSense, 2020-10-03:
There was an Android case involving use of filters that similarly had a **null pointer dereference**.  Have you seen https://github.com/IntelRealSense/librealsense/issues/5424 please?

joeyfedor, 2020-10-03:
Hi Marty,

What he describes is my exact use case unfortunately their is no mention of
his resolution steps.

MartyG-RealSense, 2020-10-03:
Hopefully he will be able to provide you with useful guidance, as he is helpful and knowledgeable.  Good luck!

MartyG-RealSense, 2020-10-12:
Hi @joeyfedor   Do you still require assistance with this case, please?   Thanks!

joeyfedor, 2020-10-12:
Hi Marty, thank you for reaching out. Yes, the actions that the other user
had me take did not resolve the issue. I could use some support.

MartyG-RealSense, 2020-10-12:
I reviewed the case from the beginning but could not find an obvious cause of the problem.  Considering that you are using SDK **2.39.0**, would it be possible please to update your camera to the firmware version that is recommended for this SDK, which is **5.12.8.200**

Doing so should eliminate whether or not the firmware is a factor in the tombstoning problem.

MartyG-RealSense, 2020-10-20:
Hi @joeyfedor   Have you had the opportunity to upgrade your camera firmware to test whether it improves your situation please?   Thanks!

joeyfedor, 2020-10-20:
Hi @MartyG-RealSense,

I just updated the firmware to 5.12.8.200 and tried again, I am getting this tombstone now: 

`_dump64: performing dump of process 1513 (target tid = 1513)
2020-10-20 07:47:44.084 16497-16497/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2020-10-20 07:47:44.084 16497-16497/? A/DEBUG: Build fingerprint: 'Nokia/Daredevil_00WW/DDV_sprout:10/QKQ1.191014.001/00WW_2_27D:user/release-keys'
2020-10-20 07:47:44.084 16497-16497/? A/DEBUG: Revision: '0'
2020-10-20 07:47:44.084 16497-16497/? A/DEBUG: ABI: 'arm64'
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG: Timestamp: 2020-10-20 07:47:44-0500
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG: pid: 1513, tid: 1513, name: xample.optiflow  >>> com.example.optiflow <<<
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG: uid: 10136
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x4791a
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x0  0000000000000000  x1  000000000000000d  x2  00000000000000e5  x3  000000783e6406d0
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x4  0000000000000001  x5  0000000000000000  x6  00000078512f30e4  x7  0000000000000010
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x8  0000000000023c8d  x9  0000000000000000  x10 00000078511335f8  x11 0000000000000010
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x12 000000785109f544  x13 000000000000000c  x14 0000005a48833000  x15 0000000000000027
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x16 0000007851523a80  x17 00000078511334ec  x18 0000000000000a00  x19 000000783e6406d0
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x20 00000000000000e5  x21 000000000000000d  x22 0000007851487258  x23 00000078514873a8
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x24 000000793fb45020  x25 000000793fb45020  x26 0000000013241ab8  x27 0000000012dd3600
2020-10-20 07:47:44.096 16497-16497/? A/DEBUG:     x28 0000000013241b60  x29 0000007fd10476c0
2020-10-20 07:47:44.097 16497-16497/? A/DEBUG:     sp  0000007fd1047520  lr  0000007851133608  pc  0000007851133610
2020-10-20 07:47:44.877 1541-1937/? D/WifiClientModeImpl: fetchRssiLinkSpeedAndFrequencyNative rssi=-46 TxLinkspeed=433 freq=5240 RxLinkSpeed=433
2020-10-20 07:47:44.877 1541-1937/? V/WifiConfigManager: Updating scan detail cache freq=5240 BSSID=02:6a:e3:8b:59:4c RSSI=-40 for "We-Fee"WPA_PSK
2020-10-20 07:47:44.882 2928-4693/? D/QCNEJ/WlanStaInfoRelay: Received action: android.net.wifi.RSSI_CHANGED
2020-10-20 07:47:44.882 1541-3027/? I/WifiService: getCurrentNetwork uid=10025
2020-10-20 07:47:44.882 2928-4693/? I/QCNEJ/NativeHalConnector: -> SND notifyWlanStaStatusChanged([WlanStaInfo]: wifiSwitchState = 1 rssi = -46 ssid = We-Fee bssid = 02:6a:e3:8b:59:4c dnsInfo = 192.168.0.1;0.0.0.0;0.0.0.0;0.0.0.0; freqBand = _5GHz countryCode = us[RatInfo]: networkType = 1 subType = 101 networkState = CONNECTED netHdl = 441492361229 ipAddrV4 = 192.168.0.18 ipAddrV6 =  ifNameV4 = wlan0 ifNameV6 =  slotIdx = 0 isAndroidValidated = true) timeStamp = 2020-10-20 07:47:44.882
2020-10-20 07:47:44.884 1541-1937/? D/WifiClientModeImpl:  L2ConnectedState screen=on 18 0 "We-Fee" 02:6a:e3:8b:59:4c rssi=-46 f=5240 sc=60 link=433 tx=41.0, 2.0, 0.0  rx=18.6 bcn=23106 [on:1524 tx:9 rx:8 period:3056] from screen [on:109803 period:162033] score=60
2020-10-20 07:47:45.004 16497-16497/? A/DEBUG: backtrace:
2020-10-20 07:47:45.004 16497-16497/? A/DEBUG:       #00 pc 00000000007a9610  /data/app/com.example.optiflow-NxfZPtOvDbPqbm4Hn14KBQ==/base.apk!librealsense2.so (offset 0x39bb000) (librealsense::depth_frame::get_distance(int, int) const+292) (BuildId: e970657889426ccbd623926dc7609b6bbd5cfc75)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #01 pc 00000000007a9010  /data/app/com.example.optiflow-NxfZPtOvDbPqbm4Hn14KBQ==/base.apk!librealsense2.so (offset 0x39bb000) (rs2_depth_frame_get_distance+236) (BuildId: e970657889426ccbd623926dc7609b6bbd5cfc75)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #02 pc 0000000000712b60  /data/app/com.example.optiflow-NxfZPtOvDbPqbm4Hn14KBQ==/base.apk!librealsense2.so (offset 0x39bb000) (Java_com_intel_realsense_librealsense_DepthFrame_nGetDistance+56) (BuildId: e970657889426ccbd623926dc7609b6bbd5cfc75)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #03 pc 000000000200dbf8  /memfd:/jit-cache (deleted) (art_jni_trampoline+168)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #04 pc 0000000002024cfc  /memfd:/jit-cache (deleted) (com.intel.realsense.librealsense.DepthFrame.getDistance+60)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #05 pc 00000000020260c8  /memfd:/jit-cache (deleted) (com.example.optiflow.RealSenseVideo$1.run+3000)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #06 pc 00000000020751a4  /memfd:/jit-cache (deleted) (android.os.Handler.handleCallback+68)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #07 pc 00000000020d66c0  /memfd:/jit-cache (deleted) (android.os.Handler.dispatchMessage+64)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #08 pc 00000000020faba0  /memfd:/jit-cache (deleted) (android.os.Looper.loop+1216)
2020-10-20 07:47:45.005 16497-16497/? A/DEBUG:       #09 pc 000000000013663c  /apex/com.android.runtime/lib64/libart.so (art_quick_osr_stub+60) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
[... cut ...]
2020-10-20 07:47:45.007 16497-16497/? A/DEBUG:       #43 pc 00000000000c282c  /system/lib64/libandroid_runtime.so (android::AndroidRuntime::start(char const*, android::Vector<android::String8> const&, bool)+776) (BuildId: 5f368f14b73392b022e30002b81606c8)
2020-10-20 07:47:45.008 16497-16497/? A/DEBUG:       #44 pc 00000000000035b0  /system/bin/app_process64 (main+1376) (BuildId: 1a86d090ff0d3dd5f5a671601a88a247)
2020-10-20 07:47:45.008 16497-16497/? A/DEBUG:       #45 pc 000000000007e898  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+108) (BuildId: 91cd22218589ba7b9ea39ce46bc68dff)`

Thank you
Joey

MartyG-RealSense, 2020-10-20:
It is admittedly a tough problem as I do not have the knowledge to diagnose RealSense Android tombstones.  A member of the RealSense community who is an Android expert though offered a patch for crash problems.  It has not been integrated into the SDK, but the script that he links to in his comment may provide some useful insight.

https://github.com/IntelRealSense/librealsense/issues/4214#issuecomment-506802514

closed 2020-10-30 by RealSenseCustomerSupport