realsenseai/librealsense #7485: pip install pyrealsense2 for python 3.8 (and later)

TDHster, 2020-10-02:
pyrealsense2 installing via pip3 install pyrealsense2 requre Python :: 3.7

But on current distributeves like latest ubuntu (20.04lts) exist python 3.8.2 

Please, update package for this way install.

ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
$ uname -a
Linux hp 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

MartyG-RealSense, 2020-10-02:
Hi @TDHster   A feature request for a pip install package for Python 3.8 already exists but there is no further information currently available about future plans for pip package versions. In the meantime, the Python wrapper can be built from source code for Python 3.8 instead of using pip.

TDHster, 2020-10-09:
I really hope that this package will be made. 
Current for default Ubuntu setup (20.04) it very hard way to build from source - getting many not evident errors for needed packages. Install via pip incomparably easier.

closed 2020-10-09 by TDHster

MartyG-RealSense, 2020-10-09:
Thanks very much for the feedback @TDHster !