realsenseai/librealsense #7486: Newer CPU required?

nadight, 2020-10-02:
| | |
|---|---|
|**librealsense**|2.39.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2020-10-02:
Hi @nadight   Information about this error can be found at the link below:

https://github.com/IntelRealSense/librealsense/issues/7080#issuecomment-671807292

The standalone version of the RealSense Viewer mentioned in the link can be downloaded from the **Assets** list at the bottom of the information listing for each SDK version on the SDK **Releases** page.  

You should launch the standalone Viewer from the location that you downloaded it to, or create a new desktop shortcut.  This is because the existing desktop shortcut for the Viewer will continue to point to the version with the CPU requirements.

https://github.com/IntelRealSense/librealsense/releases/

[image]

dorodnic, 2020-10-06:
Closing as duplicate

closed 2020-10-06 by dorodnic