realsenseai/librealsense #7487: Having Issues with D455 with the ROS v2.38.1 wrapper

Sal-Hussain, 2020-10-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (v2.38.1) | 
| Operating System & Version |   {Ubuntu 18.04.5 Lts| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.39.9 }                          | 
| Language                            |  {ROS/C++ }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, I'm having issues with connecting to the D455 device with ROS(Melodic) with Ubuntu 18.04.5. I'm using the command roslaunch realsense2_camera rs_camera. It detects the device and everything. However, I keep getting this error: [140031569065728] (backend-v4l2.cpp:1032) Frames didn't arrived within 5 seconds once it says RealSense node is up. I'm not sure why this is happening. I updated the device firmware to v5.12.08.200 and even reset the device using the realsense_viewer. The device does stream data through the realsense_viewer application. I have tried everything and I cannot seem to find a solution. Please, any sort of guidance would be greatly appreciated! 
[image]

MartyG-RealSense, 2020-10-03:
Hi @Sal-Hussain   The RealkSense ROS wrapper and librealsense SDK programs such as the RealSense Viewer operate independently and making changes to the options in one does not affect performance in the other.   

If you would like to perform a hardware reset during the roslaunch process then please add the instruction below to your roslaunch instruction to see whether it improves the problem that you are experiencing:

**initial_reset:=true**

Sal-Hussain, 2020-10-03:
Hi @MartyG-RealSense this worked! it's funny how only one option needed to be toggled to make this work. Thank you so much!

closed 2020-10-03 by Sal-Hussain

MartyG-RealSense, 2020-10-03:
Great news, thanks for the update!  Yes, **initial_reset:=true** is a good place to start when having RealSense ROS problems at launch, because it is quick to try and sometimes it does make an immediate difference.