realsenseai/librealsense #7492: ext_list.hpp file

nawara72, 2020-10-05:
Hi When I am trying build librealsense with openvino, I get the following error

Cannot find source file:
    C:/Program Files (x86)/IntelSWTools/openvino/inference_engine/src/extension/ext_list.hpp

I have installed openvino according to intel website and have ran most of its samples. I can't find the ext_list file in openvino folder.
Would appreciate your help
Thanks

MartyG-RealSense, 2020-10-05:
Hi @nawara72   If you are using the RealSense OpenVINO examples from the OpenVINO wrapper page, they were compiled for OpenVINO version **2019.3.379**.  Are you using that 2019 OpenVINO version or a more recent 2020 version please?

nawara72, 2020-10-05:
Hi Marty
Thanks for your quick reply. I am using openvino 2020.4.287, so I guess that's the problem. I will try to install the older version 2019.3.379 unless you have any ideas on how to build librealsense with the more recent versions of openvino?

Thanks again

MartyG-RealSense, 2020-10-05:
As far as I know, at the time of writing this the best recommendation would be to use OpenVINO 2019.3.379  if possible.  

Having said that, an OpenVINO user who had problems with this file in OpenVINO 2020 created workarounds by making a series of edits described in the link below.

https://github.com/fritzboyle/openvino-with-fpga-hello-world-face-detection/issues/6

The RealSense team members responsible for OpenVINO are aware of the 2019 / 2020 issue.

nawara72, 2020-10-05:
Great will check that link out. Thanks for all your great help :-)

nawara72, 2020-10-07:
Hi Marty

I was able compile the librealsense with openvino 2019.3.379. However when I open the viewer I only see the "decimation filter" under "post processing" but I can see "face detectio: openvino". Any help would be appreciated

Thanks

MartyG-RealSense, 2020-10-07:
For the RGB post-processing options in the Viewer, 'Decimation Filter' and 'Face Detection: OpenVINO' are the only options available.  So what you are seeing is correct.  

The full range of post-processing filters are under the Post-Processing section of the **Stereo Module** options but these will not affect the RGB image.

nawara72, 2020-10-07:
Sorry Marty I meant I can not see the face detection option Thanks

nawara72, 2020-10-07:
I think I am missing some dlls, I just added some dlls from openvino and now I see the face detection option, but it is not really doing anything on the rgb image when it is enabled

nawara72, 2020-10-07:
I don't know if this might help, I tried running rs-face-vino
I got
rs-face-vino.exe
[ERROR] Unknown exception caught: Failed to create plugin C:\Program Files\librealsense-2.39.0\build\win10_vs19\Release\MKLDNNPlugind.dll for device CPU
Please, check your environment
Cannot load library 'C:\Program Files\librealsense-2.39.0\build\win10_vs19\Release\MKLDNNPlugind.dll': 126 from cwd: C:\Program Files\librealsense-2.39.0\build\win10_vs19\Release

I check "MKLDNNPlugind.dll" exists in 'C:\Program Files\librealsense-2.39.0\build\win10_vs19\Release wher rs-face-vino is

Thanks

MartyG-RealSense, 2020-10-07:
OpenVINO face detection in the Viewer was introduced in SDK version **2.36.0**.  Furthermore, the OpenVINO functionality is only in the version of the Viewer that is bundled with the SDK.  It is not included in the standalone .exe version of the Viewer installed from the **Assets** list on the Releases page.

If you are using SDK 2.36.0 or newer and installed it using the 700 mb-ish SDK installer file so that you have the OpenVINO-enabled Viewer, a red box should appear around the head when you point the camera at your face if the function is working correctly.

nawara72, 2020-10-07:
I just compiled librealsense-2.39.0 and used the realsenseviewer that was built.

MartyG-RealSense, 2020-10-07:
I have not personally encountered this error, though the Intel OpenVINO forum has.  Scroll down to the final comment on the link for a suggested solution.

https://community.intel.com/t5/Intel-Distribution-of-OpenVINO/MKLDNNPlugin-loading-error-in-Deep-learning-tool/td-p/1148855

MartyG-RealSense, 2020-10-07:
I re-read your case from the beginning.  I wonder if the conflict is occurring **because** you built OpenVINO.  The OpenVINO functionality in the Viewer is not connected to the OpenVINO Toolkit as far as I know but is a standalone function that does not require OpenVINO to be installed.

nawara72, 2020-10-07:
The face-detection from the SDK i working perfectly. Though I would like get it work from the built librealsense.
When I use the cmake-gui to generate the librealsense.sln. It does ask for "INTEL_OPENVINO_DIR" so I am assuming it requires an installed openvino

I will look at the previous link you sent, it might fix both the rs-face-vino and the realsense-viewer issue built throu librealsense
Thanks

MartyG-RealSense, 2020-10-14:
Hi @nawara72   Do you require further assistance with this case, please?   Thanks!

nawara72, 2020-10-14:
Hi Marty
I wasn't able to get the face detection to work in the realsense viewer that I compiled but as I mentioned the face_detection from the SDK works, which is all i need for now. So thanks for all your help

closed 2020-10-14 by nawara72