realsenseai/librealsense #7494: bag file size

lghasemzadeh, 2020-10-05:
Is there any way to reduce the size of bag file.
I can't use SSD and I don't want to reduce resolution or FPS, because my model will be affected.
So I want to record videos and then reduce the size.
Is it possible?

MartyG-RealSense, 2020-10-05:
Hi @lghasemzadeh   If you record the bag in ROS instead of librealsense then you can use a bag function called **split** to reduce bag size.  You can set a maximum time or file size, and when that maximum limit is reached ROS will automatically close the recording and start recording a new bag file.  So you end up with a set of sequential bag files instead of a single one with a huge file size.

https://github.com/IntelRealSense/librealsense/issues/2424#issuecomment-423284173

lghasemzadeh, 2020-10-05:
Hi Marty,
But at the end they have the same size, isn't it?
having a single (1hour) bag file or having 4 (15min) files should be in same size at the end.

MartyG-RealSense, 2020-10-05:
The process should be the same as recording a single large bag except the file size of each file is smaller because they are divided into a set of multiple small bags.  But yes, the **overall total space** consumed on the storage device should be similar.  

Depending on whether you need to access the bag data afterwards or just replay it as a video like an mp4, if you record in ROS then you could use a ROS tool in the **bag_tools** bag editing toolset called **make_video.py**.  It uses **ffmpeg** to extract a video.

http://wiki.ros.org/bag_tools#make_video.py

lghasemzadeh, 2020-10-05:
yes. my problem is the overall total space.
I want both replay/watch the video and doing some post-processing and running algorithm over recorded video/frames.
Since I have already collected my data/videos with realsense there is no chance to do it again by ROS, I think there is no solution for me then.

MartyG-RealSense, 2020-10-05:
Yes, I can see your problem if you have already recorded the bags and cannot re-do the data.  Would it be possible to store the data on a high-capacity USB flash storage stick that acts like a drive letter?

lghasemzadeh, 2020-10-05:
No, in our company we are not allowed to use external SSD or USB flash.
But if I use ROS for recording videos instead of realsense in future data generation , I can use make_video.py to extract video (like MP4) from stream. is it correct? or first I need to record the bag file with ros and then convert them to video (mp4)? 
which of course MP4 takes lower space that bag file.
If I will be able to record the streams directly as video (mp4) not bag, would be great.

MartyG-RealSense, 2020-10-05:
The instruction notes for **make_video.py** suggest that the video is extracted from a bag file, implying that the bag has to be recorded first before make_video.py can convert it to a video file.

If data security is a concern for your company, perhaps a central **server** could be set up in a secure location, with individual computers saving and loading the bag files to and from the server's hard drive over a network connection.  But that may be considered by the company to be the same kind of device as an external SSD.

If you have the opportunity to record new data in future then using **depth image compression by colorization** would reduce the file size.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

lghasemzadeh, 2020-10-06:
I am not working on depth stream.
I just need RGB and IR streams. and as far as I understood 'depth image compression by colorization' is just for depth stream.

MartyG-RealSense, 2020-10-06:
I recalled that a recording compression system had been added to the SDK in the past.  It is accessible in the RealSense Viewer by following the steps below:

1.  Open the RealSense Viewer and select the gear-wheel icon in the top corner of the Viewer window.  Select **Settings** from the gear-wheel's menu.

[image]

2.  On the Settings window, select the **Playback & Record** option on top of the window and left-click on the circle beside the **Always Compress** option to enable it.  Then left-click on the **Apply** button at the bottom of the window to save the change.

[image]

closed 2020-10-20 by MartyG-RealSense