realsenseai/librealsense #7495: ”L515“  Refer to the labview sample

cja-515, 2020-10-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.0.0 | 
| Operating System & Version |  Win 10 Pro  -18362.1016 | 
| Platform                                 | PC|
| SDK Version                            |   Intel RealSense  SDK 2.0 (v2.38.1)                     | 
| Language                              |   C++                | 

### Issue Description
Hi, I want to get point cloud data in Labview many times, but it takes too long to get the existing dll, so I want to package a dll by myself. I wrote a sample program in VS with reference to the labview routine, but it can’t  obtain the depth map normally, The display window always closes directly，are there any steps missing?

using namespace std;
using namespace cv;
const auto window_name = "Display Image";
int main(int argc, char * argv[]) try
{

	namedWindow(window_name, WINDOW_AUTOSIZE);
	rs2_error** error = nullptr;
	int i = rs2_get_api_version(error);
    rs2_context* context = rs2_create_context(i, error);
	rs2_device_list  *dev_list  = rs2_query_devices(context, error);
	int j = rs2_get_device_count(dev_list,error);
	rs2_device *dev = rs2_create_device(dev_list,0,error);
	rs2_sensor_list *sensor_list = rs2_query_sensors(dev, error);
	 int sensor_count = rs2_get_sensors_count(sensor_list,error);
	 rs2_sensor *depth_sensor = rs2_create_sensor(sensor_list, 0, error);
	 rs2_sensor *color_sensor = rs2_create_sensor(sensor_list, 1, error);
	 rs2_stream_profile_list* depth_sream_list = rs2_get_stream_profiles(depth_sensor,error);
	 rs2_stream_profile_list *color_sream_list = rs2_get_stream_profiles(color_sensor, error);
	 int depth_stream_count = rs2_get_stream_profiles_count(depth_sream_list,error);
	 int color_stream_count = rs2_get_stream_profiles_count(color_sream_list, error);
	 const rs2_stream_profile *depth_stream_profile = rs2_get_stream_profile(depth_sream_list, 12, error);
	 const rs2_stream_profile *color_stream_profile = rs2_get_stream_profile(color_sream_list,18, error);
	 rs2_open(depth_sensor, depth_stream_profile, error);
	 rs2_open(color_sensor, color_stream_profile, error);
	 rs2_frame_queue*  depth_frame_queue = rs2_create_frame_queue(10, error);
	 rs2_frame_queue*  color_frame_queue = rs2_create_frame_queue(10, error);
	 rs2_start_queue(depth_sensor, depth_frame_queue,error);
	 rs2_start_queue(color_sensor, color_frame_queue, error);
	 while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
	 {
		auto depthframe = rs2_wait_for_frame(depth_frame_queue,5000, error);
		const int w = rs2_get_frame_width(depthframe,error);
		const int h = rs2_get_frame_height(depthframe, error);
	  Mat image(Size(w, h), CV_8UC3, (void*)rs2_get_frame_data(depthframe,error), Mat::AUTO_STEP);

		imshow(window_name, image);
		rs2_release_frame(depthframe);
		}
	rs2_close(depth_sensor,error);
	rs2_close(color_sensor, error);
	rs2_delete_device(dev);
	rs2_delete_device_list(dev_list);
	rs2_delete_context(context);
	system("pause");
	return EXIT_SUCCESS;
}`
[image]

RealSenseSupport, 2020-10-05:
Hi @cja-515 

We will need to investigate although there currently is minimal support for Labview using L515.

dorodnic, 2020-10-06:
I see that the code you wrote is partly C++ (due to auto)
Have you tried rs-imshow example? I think using C++ API for this task would be easier.

cja-515, 2020-10-09:
Hi    @dorodnic  
Yes, I have tried the rs-imshow example, but the image is taken through the pipeline, I can not choose its image resolution, I can choose the image resolution for the labview example, but  the speed of taking pictures is too slow, so I want to try how long it takes to take pictures in C++ with the programming idea of Labview. If the speed of taking pictures in C++ is fast, I will pack a dll for taking pictures by myself. Could you please help me to see if there is an error in my program? Is it possible to select the resolution of the image when the image is taken by pipeline?

cja-515, 2020-10-09:
When can L515 support labview? @RealSenseSupport

RealSenseSupport, 2020-11-13:
Hi @cja-515 

We cannot comment on support timeline at this stage regarding support for labview.  Using the suggestion by dorodnic would be best in this situation.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-08 by RealSenseSupport

om262, 2021-03-11:
Hi @RealSenseSupport 

Just to resurrect the dead; has there been any movement regarding LabVIEW support for the L515?