realsenseai/librealsense #7496: Calibration tool GUI: Error (9998): Unknown

roelofvandijk, 2020-10-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Win 10 /   | 
| Platform                                 | PC |
| SDK Version                            |2.38.1                          | 
| Language                            |  Calibration tool GUI/CLI                         | 
| Segment			| Calibration                       | 

### Issue Description

I have a D415 with FW 05.12.07.100. The images display fine in the viewer, and the calibrations inside the RealSense viewer all work. I use the original cable and the RealSense viewer and rs-enumerate-devices recognize the camera on USB 3.2.

However, the Dynamic Calibration Tool (v2.11.0.0) does not work at all (in GUI and CLI).

#### Output from the CLI:

```
Serial Number: censored
Device Name: Intel RealSense D415
Device PID:  0AD3
FW version:  05.12.07.100

Calibration Type: Targeted
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Target images: 15
Target and device orientation aligned: No
Skip rgb calibration: No
Initializing GL ...
display size: 1920,1080 with aspect ratio 1.77778
window size: 1440,810 with aspect ratio 1.77778
Launching graphical view ...
Streamming Resolution: 1280x720 at 30(fps)
disable AUTO_EXPOSURE_PRIORITY to ensure frame sync between rgb and depth/left/right sensors

Error (9998): Unknown
```

#### Output from the GUI (program remains in the starting screen, no images).
[image]

I have reinstalled the firmware, the SDK, the calibration tool, and the windows drivers to no avail. I have also tried the newest firmware (5.12.8.200).

MartyG-RealSense, 2020-10-05:
Hi @roelofvandijk   As you already have the Dynamic Calibrator tool installed, a tool called **CustomRW** should also be installed.  Could you try launching the tool with the command below please to perform a **Gold Reset** of the camera.  This resets the camera to its factory-new configuration and may also fix the problem of a corrupted 'calibration table' inside the camera:

**Intel.Realsense.CustomRW -g**

roelofvandijk, 2020-10-05:
Thank you for you reply.

After uninstalling everything, including the RealSense Depth and RGB camera in the device manager and deleting the driver, and then reconnecting the RealSense, the Calibration works.

I think it is highly unlikely that both camera's (D415 and D435) were both corrupted and think it is an issue with the Windows driver. I'll try to reproduce.

roelofvandijk, 2020-10-05:
I have reproduced the problem:

Clean PC:
- After installing only the Calibration tool: Calibration works
- After installing the SDK (2.38.1) as well: Calibration works
- After installing the D400 UWP Driver (6.1.160.21, DMFT_Version 6.1.160.21.zip from [here](https://downloadcenter.intel.com/download/29827/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10?product=85364): **Calibration is broken with the above error**
- After removing the driver in device manager: Calibration works
-

MartyG-RealSense, 2020-10-05:
You do not need to have the UWP driver installed to use RealSense.  It is used for providing RealSense support when attempting to interface with products that use Microsoft's **Universal Windows Platform (UWP)**, such as the Microsoft Hololens headset.  For this reason, most RealSense users do not need it.  There was a case reported in the past week where a user of the RealSense Unity wrapper found that their project crashed if the UWP driver was installed.  Removing the UWP driver fixed the problem.

roelofvandijk, 2020-10-07:
That makes sense, I just followed the recommended table here:
https://dev.intelrealsense.com/docs/firmware-releases

Thanks for your help.

MartyG-RealSense, 2020-10-14:
Hi @roelofvandijk   Do you require further assistance with this case, please?   Thanks!

roelofvandijk, 2020-10-14:
Hello @MartyG-RealSense, thanks for asking. I currently don't need any assistence as it works without the UWP driver, but I do consider this a bug.

MartyG-RealSense, 2020-10-14:
A bug report for the issue with this particular UWP driver version has already been filed with Intel.  If you do not need further assistance then I will close this case.  Thanks again!

closed 2020-10-14 by MartyG-RealSense

crisp7777, 2021-12-27:
Thanks for the hint regarding removing UWP-driver for calibration. The bug is still present in newer release UWP 6.1.160.22. I had the same issue where I got the 9998 error during calibration attempt but when I removed the UWP-driver it worked like a charm.