realsenseai/librealsense #7502: D455 External Sync is not supported for color camera

sam598, 2020-10-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |  Raspberry Pi OS  | 
| Kernel Version (Linux Only)    |   5.4.65-v7l+  | 
| Platform                                 | Raspberry Pi |
| SDK Version                            | 2.38.1   | 
| Language                            |  python   | 

### Issue Description

When running the the D455 in "genlock" mode the external signal only triggers the depth camera. The color camera runs out of sync and at the full frame rate.

Trying to call: `set_option(rs.option.inter_cam_sync_mode, 4)` on the color sensor returns the error: `object doesn't support option #42`

This is the same behavior as the D435. It is expected on the D435 because the color camera is a rolling shutter and separate from the global shutter stereo cameras.

However the color camera on the D455 is a global shutter, and apparently the exact same sensor as the two infrared cameras. In theory this camera should not have the same limitation. Multiple times this whitepaper states this feature is "supported for Intel RealSense Global Shutter cameras":
https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

Is this something that can be fixed in a software/hardware update? Or is it a hardware limitation where the color sensor on the D455 still separate from the infrared sensors like on the D435?

MartyG-RealSense, 2020-10-06:
Hi @sam598   I would speculate that since color is not usually synced to depth in the Extenal Synchronization (Genlock) mode due to the hardware limitation of the color rolling shutter on camera models equipped with Global Shutters that preceded the D455 (e.g D435 and D435i) then D455-specific support for genlocking the color sensor is not currently part of the External Synchronization function.

@dorodnic would be better able to comment on whether D455-specific support for color genlocking is likely to be added in the future.

sam598, 2020-10-07:
I can also confirm that this behavior is the same in v2.39.0 of the SDK and firmware 5.12.8.200

In the latest firmware I can get a "genlocked" color image from the left IR sensor, but this is not the ideal color camera.

sam598, 2020-10-09:
Sorry to ping you again @dorodnic, but if this is a hardware limitation like with the D435 and a firmware/SDK update is completely out of the question, it would be better to know sooner than later so I can look for an alternate solution.

agrunnet, 2020-10-09:
There is not HW limitation. But we have not enabled it yet.

sam598, 2020-10-09:
Thanks @agrunnet that is great news. Looking forward to it!

urock, 2020-10-16:
Guys, looking forward for this feature too!

hocop, 2020-10-19:
This would be great to have this feature enabled! @agrunnet

MartyG-RealSense, 2020-10-27:
I will add a note to keep this case open for the time being as there is strong interest in the feature mentioned by @agrunnet

sam598, 2020-10-27:
Thanks @MartyG-RealSense 

@agrunnet I know and greatly appreciate that the RealSense team is moving as fast as they can to add new features every month. Is there any chance this will make it into next month's release?

MartyG-RealSense, 2020-11-04:
Adding a reminder for @agrunnet about the question of @sam598 above.   Thanks!

s-t-e-p-a-n-o-v, 2020-11-08:
Hi @agrunnet @MartyG-RealSense !
Recently we received D455 cameras and looking for this feature too.
We are working with x86 platform but it does not matter in this case.
Ability to sync RGB sensor was main reason for switch D435 to D455.
Could you provide ETA for this feature ?
Thanks!

MartyG-RealSense, 2020-11-08:
Hi @s-t-e-p-a-n-o-v   I do not have that ETA information.  @agrunnet may do.  Thanks for your patience.

agrunnet, 2020-11-10:
Request noted and we are working on it. I do not have a release date yet.

FrankCreen, 2020-11-11:
> Required Info
> Camera Model	D455
> Firmware Version	5.12.7.100
> Operating System & Version	Raspberry Pi OS
> Kernel Version (Linux Only)	5.4.65-v7l+
> Platform	Raspberry Pi
> SDK Version	2.38.1
> Language [...]

Hi ！I test the code below with a 75FPS, could you show your result? Thanks very much!

## Test Program
```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

    //Sensor Settings
    auto sensor1 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor1.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

MartyG-RealSense, 2020-11-11:
Hi @FrankCreen    Please post your question as a new issue using the **New Issue** button at the link below and do not post it on multiple D455 cases if the question is unrelated to that case.   Thank you.

https://github.com/IntelRealSense/librealsense/issues/

Alamoris, 2020-11-17:
Really looking forward to when this functionality will work.

MartyG-RealSense, 2020-11-26:
As there is not a release date yet for the feature but it is being worked on, I wil close this case in the meantime.

sam598, 2020-11-26:
The issue has not been fixed and has no release date, why was it closed?

MartyG-RealSense, 2020-11-26:
This case is not linked to the feature's development and so it does not need to be kept open - work on the feature will continue even if this is closed.

urock, 2020-11-26:
This is long awaited feature I am keeping track of. Now when you have closed this issue how should I (and other developers) know  that it least this feature has been selected for development?

MartyG-RealSense, 2020-11-26:
You would not automatically receive a notification of a new feature's release.  So the open / close status would not make a difference.  You can check the release notes of new librealsense releases for details of what has been added to it

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes

Closed discussions remain findable by internet search for developers searching for a subject, such as as external sync for color camera.

sam598, 2020-11-26:
@MartyG-RealSense as the one who opened the issue, I will happily close it once the release notes say that the feature has been added (if you would please re-open it).

As you have said yourself there is no way for people subscribed to the thread to be notified otherwise.

MartyG-RealSense, 2020-11-26:
Hi @sam598   As the creator of this discussion, I will respect your wish for the discussion to be open.  I have added an **Enhancement** tag to this discussion as a reminder for it to be kept open.  There is no time schedule that I can provide for release of the feature though.

sam598, 2020-11-26:
Thanks @MartyG-RealSense I greatly appreciate it. I have purchased 12 of these cameras for a project that is on hold until this fix is released, so I can assure you I am checking for updates regularly.

s-t-e-p-a-n-o-v, 2020-12-22:
Hi! I would like to up this thread.
Is there any news about ETA for now ?

MartyG-RealSense, 2020-12-22:
Hi @s-t-e-p-a-n-o-v   I do not have that information unfortunately.

MartyG-RealSense, 2020-12-29:
Adding a note to keep this case open for a further time period.

s-t-e-p-a-n-o-v, 2021-02-18:
Hi @sam598 !

Saw your reply here https://github.com/IntelRealSense/librealsense/issues/8197#issuecomment-763904671

"
Genlock through external sync signal (modes 4-255):
D415 - No Genlock support. All cameras are rolling shutter.
D435 - Genlock support for infrared/depth cameras. No Genlock support for rolling shutter color camera.
D455 - Genlock support for infrared/depth cameras. Genlock support for color cameras are planned but no ETA.
One RealSense camera generates a master signal, others are slaved to it (mode 1 and 2):
D415 - Full support.
D435 - Rolling shutter color camera may not be in sync with Global Shutter infrared cameras (this may have changed).
D455 - Full support.
"

Does it mean that color camera of D455 is in sync when master-slave mode is applied (mode 1 and 2) ?

s-t-e-p-a-n-o-v, 2021-06-17:
Hi!

@MartyG-RealSense, is there any updates on this ?

MartyG-RealSense, 2021-06-17:
No sorry @s-t-e-p-a-n-o-v   I do not have any information to provide on this subject.

sam598, 2021-06-17:
@s-t-e-p-a-n-o-v 

In some tests I ran with recent a firmware and SDK version, it appears now that the color and depth cameras of the D455 will be synced in modes 1 and 2 (master and slave). https://github.com/IntelRealSense/librealsense/issues/8958#issuecomment-833060193

Mode 0 'default', Mode 3 'full slave', and Modes 4+ 'genlock' did not run in sync in my tests.

In a previous comment RealSense support said that the D455's color camera would not run in sync in mode 2, so this would seem to contradict that. There has been no official word from the RealSense team about this yet. @MartyG-RealSense since the devices are working differently from what Intel has said, it would be great to get clarification.

MartyG-RealSense, 2021-06-18:
@sam598  I do not have any information to share on the subject of D455 RGB multicam sync.

s-t-e-p-a-n-o-v, 2021-09-02:
Hi! I would like to ask status again ? Will it be realized at all according recent news ?

MartyG-RealSense, 2021-09-02:
Hi @s-t-e-p-a-n-o-v   I am now able to provide information about the future of RealSense, as an 'End of Life' notice has been released by Intel.  Although some RealSense models are being discontinued, the 400 Series stereo depth cameras are continuing.  The official PDF of the End of Life notice explaining the changes is attached below.

[PCN118463-00.pdf](https://github.com/IntelRealSense/librealsense/files/7100557/PCN118463-00.pdf)

Whilst the D455 model and the librealsense SDK are continuing, I still do not have information specifically about D455 RGB multicam sync.

MartyG-RealSense, 2021-09-18:
Hi everyone, thanks very much for your patience on this issue.  Intel's investigation of D455 hardware sync for RGB is now closed.  It concluded that when comparing a frameset of depth and RGB in order to calculate the sync, the difference between them is determined by FPS, and although sync on the D455 was slightly more shaky than the D415 camera model, it was consistently steady.

sam598, 2021-09-19:
Hi @MartyG-RealSense, I appreciate you following up on this issue.

I have to admit I do not understand your last post. Why would Intel need to "investigate" their own product? Shouldn't they know which features the camera and SDK do and do not support? Because this is a rather long thread please let me try to simplify the initial question:

Is it possible for the D455 to synchronize (capture at the same moment in time) RGB and Depth frames when receiving an external sync signal (modes 2, 3, or 4+)?

Earlier in this thread @agrunnet said that the feature may come in an SDK update:

> There is not HW limitation. But we have not enabled it yet.

Has this feature been added? If so what is the correct way of enabling it? Does it happen automatically? Or does the sync option need to be set on the RGB sensor?

Thanks again!

MartyG-RealSense, 2021-09-19:
When a feature request is made or an issue that is possibly a bug is identified, that request or possible bug goes through a formal review process.  The outcome of this may be that an action is taken or no further action is taken for various reasons.

In this particular case relating to D455 RGB sync, the review was concluded and closed.  The decision at the outcome of the process is final.  My understanding of the above explanation is that no further action was deemed necessary because the depth-RGB sync was already found by the review process to be performing steadily on D455.

sam598, 2021-09-19:
Thanks @MartyG-RealSense. So you are saying that it is now possible to synchronize RGB and Depth frames (capture at the same moment in time) on a D455 when receiving an external sync signal (modes 2, 3, or 4+)?

MartyG-RealSense, 2021-09-20:
I do not have access to sync cabling and equipment to test the feature myself.  I recommend performing tests with your own sync setup to see what results you achieve.

sam598, 2021-09-20:
I'm sorry but I do not understand why would you @MartyG-RealSense would need to test it yourself if Intel has already "investigated" it. That makes it sound like this feature may not work.

Previously the D455 could not sync RGB and Depth from an external sync signal. Has this feature been added? This should really be a "yes" or "no" answer.

MartyG-RealSense, 2021-09-20:
I was making the point that whilst the investigation regarding whether or not to proceed with adding an RGB-depth sync feature for D455 concluded that performance on D455 was steady, I am not able to test RGB-depth sync myself and so suggested that you try it yourself to see what results you get.

Please bear in mind that I am not involved in evaluations of RealSense feature requests or bug reports. I report the outcome of the process.  The purpose of the feature request that was concluded was whether to add RGB-depth sync for D455.  It was concluded that D455 already provided steady RGB-depth sync performance.  You could take that as saying No to creating a new D455 RGB-depth sync feature because D455 RGB-depth sync is already working (and FPS is the primary determining factor in how well it works).

sam598, 2021-09-20:
I understand that you are only passing along information. But the information you are forwarding does not seem to answer the issue at hand.

As stated in the title, the issue is whether or not the D455 is capable of RGB-depth sync from an **external sync** signal. Not just RGB-depth sync on a single D455. It is not clear from your report if the D455 was in sync mode 2 (validated), 3 (placeholder), 4 (experimental), or receiving a sync signal at all.

It is also incorrect to say that FPS is the primary determining factor in whether or not sync works. Two cameras can run at the exact same frame rate, but if those cameras are not capturing the same moment in time each frame, then they are not in sync.

It is not possible to determine sync by just looking at FPS. If that is all that the investigator did, then they did not validate that the RGB and Depth cameras are in sync.

If there is any other information you can provide about the report it would be appreciated. Otherwise it is not clear that this issue has been resolved.

MartyG-RealSense, 2021-09-20:
There is nowhere further for this case to go, unfortunately.  The D455 RGB-depth sync feature request is closed and will not be re-opened or recreated, and there is no further information that can be provided.  The only possible outcome remaining is to close this case, even if you did not receive the answer that you wanted.  I do very much understand your frustration.

sam598, 2021-09-20:
> The D455 RGB-depth sync feature request is closed

It feels like you are choosing these words very specifically. By "D455 RGB-depth sync feature request" does that mean sync sync from an external or multi-camera signal? Because that is what this open GitHub issue is about.

> The only possible outcome remaining is to close this case, even if you did not receive the answer that you wanted.

It very well might be the answer to this issue, but there seems to be an unwillingness or inability to clarify.

MartyG-RealSense, 2021-09-20:
I regret that you are not satisifed, but there is nothing further that can be said or done in this case.

sam598, 2021-09-22:
If you are saying that the feature is working then why would I be unsatisfied? That doesn't make any sense.

I am simply asking for clarification so that this issue can be closed.

MartyG-RealSense, 2021-09-22:
Please test RGB-depth sync with your own D455 sync setup.  That will provide the answer about whether it now works for you.

MartyG-RealSense, 2021-10-03:
Hi @sam598  Where you able to test RGB-depth sync with your own D455 sync setup to confirm whether it now works for you, please?  Thanks!

closed 2021-10-14 by MartyG-RealSense