realsenseai/librealsense #7503: SR300 IR Glare

lghasemzadeh, 2020-10-06:
Hello,

I have a SR300 camera and when I try to capture IR streams it makes glare on the glasses or sunglasses.
How can I eliminate these glares? down you can see the screenshot
[image]

and here are some photos of my camera, it seems it doesn't have glass over lenses originally or it is broken. since when I find the camera in our lap it was like this.
[image]
[image]
[image]

I have read some similar issues but I didn't get a clear answer and I think my camera is not the original version or it is a trial version.

dorodnic, 2020-10-06:
Hi @lghasemzadeh 
Do you have alternative source of illumination in IR frequency? I think there is a way in camera controls to disable emitter and get only the pure IR image, but if you don't have IR light source, the image will be black. If you are using the camera as IR light source, you will always get this glare, because its simply a reflection of the camera emitter.

lghasemzadeh, 2020-10-06:
my questions are:
1) why this camera is different from other SR300 in appearance?
2) why it doesn't have glasses over the lenses?
3) I don't have any other IR light source, light comes from the camera. In case of putting a suitable glass on the port that emits light and eliminate the glare, IR camera won't see anything then I need to have an external IR light to cover the scene, right? how can I provide other source of IR light? what I should buy?

MartyG-RealSense, 2020-10-06:
Hi @lghasemzadeh   QS can stand for Quality Sample, also known as Engineering Sample.  In other words, pre-release test versions of a camera before they are launched and go on public sale.  This may account for why your model looks so different.   So I do not know if this particular QS model ever had front glass if it was an early test unit.

In regard to adding your own cover glass, I am mainly familiar with the 400 Series cover material, so @agrunnet may be able to suggest a suitable material for the SR300's technology.  It may end up easier and cheaper just to buy a modern SR305 camera though (which has SR300 technology inside).

RealSenseCustomerSupport, 2020-10-07:
----------------------------------------------

Hi @lghasemzadeh
What is your application? will you need both depth and IR signal, right?
SR300 is EOL and its successor is SR305 as MartyG already mentioned
We have have D4xx camera which can streaming both depth and IR at same time

1) why this camera is different from other SR300 in appearance?
 -as MartyG-RealSense said, this is not a production version

2) why it doesn't have glasses over the lenses?
 -this is not a production version and this not has adverse impact to its performance
on datasheet chapter 7.7.1 Cover material transmission  has the spec of the glass cover
https://www.intelrealsense.com/wp-content/uploads/2019/07/RealSense_SR30x_Product_Datasheet_Rev_002.pdf

3) I don't have any other IR light source, light comes from the camera. In case of putting a suitable glass on the port that emits light and eliminate the glare, IR camera won't see anything then I need to have an external IR light to cover the scene, right? how can I provide other source of IR light? what I should buy?

-(a) you can turn off the laser projector (emitter) by change the laser power to 0
Realsense-viewer ->Coded-light Depth Sensor->Controls ->Laser Power

-(b) as to the light source, you can check the light source used for normal IR surveillance camera (CCTV) in the market and place it at proper position to prevent the glare

as to wavelength 850nm definitely work but will interference the projector signal (also 850nm wavelength), as to 940nm which @agrunnet mentioned should also work

-(c) laser projector signal is required to obtain depth, so if you need depth info, you need to turn the laser projector on again, if the projector signal is blocked, you will not able to depth signal anymore

(d) when turn on the projector to get the depth, you need to turn off the external IR light if it is also 850nm

lghasemzadeh, 2020-10-07:
I just need color stream (RGB) and IR. I don't need depth info.
Especially for my research I need a powerful IR camera, which will produce high quality IR streams can make all types of sunglasses transparent as well as a long range capability. 
I am testing all procedure from beginning to the end on SR300 to find the weaknesses and challenges which we might face and find their solution and then start the real phase of the research. also as the result of them we will figure out exactly what kind/type of camera we would need.

BTW I changed the lazer power to zero, and yes the glare disappeared but as you mentioned the scene gets very dark, so there should be an external IR light source to be able to see the scene. 

Thank you @RealSenseCustomerSupport @MartyG-RealSense  for the complete and the efficient answer.

lghasemzadeh, 2020-10-09:
@RealSense-Customer-Engineering @RealSenseCustomerSupport @MartyG-RealSense
Thank you.
In our company we need to purchase suitable camera for line production.
I tried to find a contact to arrange a meeting but I wasn't successful.
would you please contact me via my email?

MartyG-RealSense, 2020-10-09:
Hi @lghasemzadeh   Could you clarify your needs please.  Is it the SR300 that you are still interested in?  And are you hoping to integrate it as a circuit board into a product made on a production line?  Thank you.

lghasemzadeh, 2020-10-09:
No, we tested SR300 and understood that it can not cover our requirements.
We will first prepare a document regarding our needs and challenges we faced, after reading the report you will come up with solutions and propose a suitable camera.
We prefer presence of at least  one of your engineers, IR specialists, and AI experts if you have.

how can I share my email address with you, or taking a responsible person's email address?
Thank you

MartyG-RealSense, 2020-10-09:
For advanced technical support such as video conferencing and site visits, a Non Disclosure Agreement (NDA) arrangement with Intel RealSense would be required.  Otherwise, forums such as this one are the main channel for technical support for customers without an NDA agreement.

If you would like to apply for an NDA agreement, please send your contact details to the email address below.  These details could include a real name, email address, company address, website address, etc.

martyx.grover@intel.com

lghasemzadeh, 2020-10-09:
I sent you an email.
Thanks

MartyG-RealSense, 2020-10-09:
I have responded.  Thank you.

closed 2020-10-12 by RealSenseCustomerSupport