realsenseai/librealsense #7506: L515 - The demo does not show anything on UNREAL

Sashi-Mee, 2020-10-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.93.0.2337 }                          | 
| Language                            |  {C++/unreal blueprint}                          | 
| Segment			|  {VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi !
I foloowed the manual to get started, and after a couple setback I managed to build the project for unreal engine IV.
It lunches fine, sees the camera in the hardware section of the inspector (the camera works fine outside of unreal) but it doesn't do anything on the cubes and there's no point cloud. I'm trying to see if there's something to setup to make it work but I can't find what to do... Could you help me please ?

(Once it works, i'd like to use the PCL to drive niagara particle systems. I did it with standard meshes, I hope this will work if yuo have any ideas / suggestions on the topic I'm all ears too! 👍 
Thanks in advance !

RealSenseSupport, 2020-10-08:
We are investigating Unreal support and L515 camera.

Sashi-Mee, 2020-10-09:
Okay thanks. It would be amazing that it'd work by next friday, as I participate in a hackaton weekend on moiton design and performing arts, and it would be a great tool to bridge reality and virtual worlds.

Thank you i andvance for the efforts ! :)

manson, 2020-12-30:
The same problem I had. The problem occurred because of default camera resolution set is incompatible with L515 camera. Besides, this resolutions differ according to USB type the camera is attached through. Please, try this parameters in Inspector parameters to make it work: 
1) For USB3 - D:1024x768x30, I:1024x768x30, C:1920x1080x30 or D:320x240x30, I:320x240x30, C:960x540x6 
2) For USB2 - D:320x240x30, I:320x240x30 ,C:640x480x30
(D - depth config, I - infrared, C - color)

At this point there is an unaccepted pull request (https://github.com/IntelRealSense/librealsense/pull/8036) that besides the converting project to UE4.26 makes camera autoconfiguration. You can try this.

RealSenseSupport, 2021-01-13:
@Sashi-Mee - Have you been able to test out the information provided by manson?

*If we don’t hear from you in 7 days, this issue will be closed.

Sashi-Mee, 2021-01-14:
Hi !
I didn’t see the replies !

I’ll burrow the camera from my friend to check this out !
Thanks! :)

RealSenseSupport, 2021-01-21:
Hi @Sashi-Mee - Have you been able to get the camera and test?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-03-22 by RealSenseSupport