realsenseai/librealsense #7507: L515 World Origin

tpwrules, 2020-10-06:
I can't determine the origin of the depth sensor of the L515 relative to the sensor body. The datasheet specifies that Z=0 is 4.5mm behind the front of the glass, but it does not specify where X=0 and Y=0 are. What is this point? Is it the center of the body? For comparison, the D400 series datasheet specifies that X=0 and Y=0 are at the center of the left imager.

MatsumotoJ, 2020-11-04:
@RealSenseSupport,
I also need this information. Please help. Thanks!

MatsumotoJ, 2020-11-05:
I find that the x y origin locates center of the body, because the extrinsic parameter of the RGB camera (obtained with get_extrinsics_to() function) indicates translation was [0.0008, 0.0141, -0.0065] (in my L515).

RealSenseSupport, 2020-12-08:
Hi @tpwrules and @MatsumotoJ 

I hope this additional information helps:

From Bottom of L515 To Depth Origin = 26.15mm

Depth origin location from X perspective is in the middle of the camera – Datum B

*If we don’t hear from you in 7 days, this issue will be closed.

MatsumotoJ, 2020-12-08:
@RealSenseSupport 
Thanks for the info!

RealSenseSupport, 2020-12-16:
Hi @MatsumotoJ 

We are looking into this.  Thanks for pointing this out.

RealSenseSupport, 2021-01-06:
Hi @MatsumotoJ and @tpwrules 

Modifying our previous update here.  The location of the depth origin is the middle of the IR transmitter which is 14mm from the RGB camera in the y direction which corresponds to the value you're getting from the extrinsic calibration report out.  And this is 31.15mm from the bottom of the camera.

*If we don’t hear from you in 7 days, this issue will be closed.

tpwrules, 2021-01-06:
Could you please be sure to update the datasheet with this information, including appropriate diagrams? Thanks.

RealSenseSupport, 2021-01-07:
Yes we will make a note of this for future datasheet revisions.

RealSenseSupport, 2021-01-19:
There has been a notation made in the datasheet.  Please see revision 003 that is posted on the website:  https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-27 by RealSenseSupport