realsenseai/librealsense #7508: L515 Global Timestamps no longer working

JeremyBYU, 2020-10-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.01.03 (latest) | 
| Operating System & Version |  Windows 10 and Linux  | 
| Kernel Version (Linux Only)    |  5.3 for Linux                                         | 
| Platform                                 | PC X64  |
| SDK Version                            |      2.39 (release and github master)                     | 
| Language                            |  RealSense Viewer                       | 
| Segment			|  Robot                    | 

### L515 Global Timestamps no longer working

I have noticed that the L515 Global Timestamps are no longer working on 2.39 but were working on 2.38. I have tested this on Linux X64 Ubuntu 18.04 and Windows 10. Attached are videos demonstrating this issue on Windows 10. The first video shows 2.38 successfully working while the second video shows 2.39 not working.  Below are screenshots as well:

2.38
[image]

2.39
[image]

These tests are performed on the same Windows 10 Machine. The only difference is the realsense-viewer executable downloaded form the releases pages. 

[L515_Global_Timesamps.zip](https://github.com/IntelRealSense/librealsense/files/5336280/L515_Global_Timesamps.zip)

Similar issues of an "inf" timestamp occurs on linux as well. I tried both the development branch and the master branch. I tired to look through any other issues reporting this or any pull requests but could not find any. Another useful piece of information may be that my d435i camera global timestamps work just fine on 2.38 and 2.39.

Just another random question: Why is the default value for global timestamps disabled for the L515 but enabled for the D4XX cameras?  Something I noticed but wasnt sure about.

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/l500/l500-color.cpp#L68
vs 
https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/ds5/ds5-color.cpp#L79

ev-mp, 2020-10-07:
@JeremyBYU hello, thanks for the feedback.
It is a bug that appeared in v2.39.0, And will be addressed in the next version

JeremyBYU, 2020-10-10:
Thanks for your quick response! Can you respond to the last part of my question also?

> Just another random question: Why is the default value for global timestamps disabled for the L515 but enabled for the D4XX cameras? Something I noticed but wasnt sure about. [...]

wangsff, 2020-10-15:
HI @ev-mp @
Could you please help figure out the "strange duplication region"? #7556 It's very important to me.
And I create a new issue about L515 align offset problem #7581 
Could you please help me solve this problem?
Thanks in advance!

RealSenseSupport, 2020-12-23:
@JeremyBYU 

Is there anything further needed on this issue?  It looks like the PR is already in the latest 2.40 LRS release.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-31 by RealSenseSupport