realsenseai/librealsense #7509: Unity3d build for x86 architechture shows no pointclouds

TarekFinke, 2020-10-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | .bag data | 
| Firmware Version                   | | 
| Operating System & Version |  Windows 10 -> Android 10| 
| Kernel Version (Linux Only)    | | 
| Platform                                 | PC -> Smartphone|
| SDK Version                            |  2.0 | 
| Language                            |  unity| 
| Segment			| AR/others| 

### Issue Description
Trying to include .bag data (I use the stairs.bag as source for the rsdevice) in unity projects works perfectly fine if I build for x86_64, but neither on x86 nor on Android builds are the pointclouds visible.

MartyG-RealSense, 2020-10-07:
Hi @TarekFinke   Have you tried it with a different bag from the sample data page such as **outdoors.bag** (the first bag in the list) to confirm whether or not the problem is with the bag file, please?

MartyG-RealSense, 2020-10-07:
Hi @TarekFinke   I wonder if the problem is that you are building for x86 architecture in the Unity build settings but the librealsense DLL files installed in your project are the x64 version.

MartyG-RealSense, 2020-10-07:
Hi @TarekFinke   I tried running the **StartHere** introductory scene of the Unity wrapper in Unity in both x86 and x64 mode to test for whether the installed librealsense DLL library files installed in the Unity project were able to work with a project whose Build Settings were set to x86 instead of x64.   It did work with both types, at least for the **StartHere** scene.

Do you get any error messages in the Unity log when running, or does it appear to run but the point cloud simply fails to appear?

TarekFinke, 2020-10-08:
Hi @MartyG-RealSense I can build in both versions, I tried the PointCloudDepthAndColor as well as PointCloudProcessingBlocks. Both only show PointCloud data with a x64 build. Also I tried with outdoors.bag, the same problem occurs.

TarekFinke, 2020-10-08:
It does run without any error messages, I just don't see any data

MartyG-RealSense, 2020-10-08:
@TarekFinke   There was a previous case where the Unity wrapper samples were not working because the library files had been built for x86 instead of x64.

https://github.com/IntelRealSense/librealsense/issues/2227

Unity has been based on x64 since Unity 5 was introduced some years ago.  And the wrapper's CMake instructions have deliberately built for x64 since the case in the link above by including **-DCMAKE_GENERATOR_PLATFORM=x64** in the CMake build instruction.  It is therefore likely that the wrapper is not intended to work with x86.

closed 2020-10-25 by MartyG-RealSense