realsenseai/librealsense #7510: No Device Connected (But realsense-viewer works)

kylelscott, 2020-10-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.0.00 | 
| Operating System & Version | Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  Linux 4.9.140-tegra | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  SDK 2.0 | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

When I try to run any of the example programs under the wrappers/python/examples directory I get an error of "No Device Connected". I know there are several issues open with this same problem, but mine differs in the fact that the camera is connected and opened when I run Intel RealSense Viewer, its only when I try to run .py scripts. Does anybody know why this might be happening

`kylel@kylel-desktop:~/librealsense/wrappers/python/examples$ python python-tutorial-1-depth.py
No device connected
kylel@kylel-desktop:~/librealsense/wrappers/python/examples$ lsusb | grep 8086
Bus 002 Device 004: ID 8086:0b3a Intel Corp. 
`

MartyG-RealSense, 2020-10-07:
Hi @kylelscott   If you installed the Python wrapper from source with CMake and are using a version of Python 3 (e,g Python 3.6, 3.7 or 3.8), you could try the **-DPYTHON_EXECUTABLE** term below in your CMake instruction if you have not done so already:

**-dpython_executable=/usr/bin/python3 **

MartyG-RealSense, 2020-10-07:
Hi @kylelscott  May I confirm please that you do not have the RealSense Viewer open when you are running the Python examples.  If you do, does it make any difference if you close the Viewer and then run the Python samples?

ev-mp, 2020-10-07:
@kylelscott , make sure you compile the python wrapper with `-DFORCE_LIBUVC` or `-DFORCE_RSUSB_BACKEND` cmake flag as mentioned in the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md).

MartyG-RealSense, 2020-10-14:
Hi @kylelscott  Do you require further assistance with this case, please?  Thanks!

kylelscott, 2020-10-14:
No I think the issue has been resolved. Thank you for your help.

MartyG-RealSense, 2020-10-14:
Okay, thanks very much for the update!  I will close this issue now.

closed 2020-10-14 by MartyG-RealSense