realsenseai/librealsense #7511: Depth frame count depends on the format asked

AMArostegui, 2020-10-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Win10 / Android 9  | 
| Platform                                 | Android  |
| SDK Version                            |  2.38.1 | 
| Language                            |  Java |                          | 

### Issue Description

In my Android device, I'm loading a bag video using the [playback example in the android wrapper folder](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/playback)

The bag video was recorded using the following settings:

RGB Camera: 640x480 15fps RGB8
Stereo: 640x480 6fps Z16

I have also added these lines to the example provided:

```
while (!mStreaming.isInterrupted()) {
	try (FrameSet frames = pipeline.waitForFrames()) {
		try (FrameSet processed = frames.applyFilter(colorizer)) {

			Frame frmColor = processed.first(StreamType.COLOR);
			if (frmColor == null)
				nullFrmColor++;        // <===== This is always zero. OK

			Frame frmDepthRgb = processed.first(StreamType.DEPTH, StreamFormat.RGB8);
			if (frmDepthRgb == null)
				nullFrmDepthRgb++;     // <===== This variable keeps increasing. OK

			Frame frmDepthZ16 = processed.first(StreamType.DEPTH, StreamFormat.Z16);
			if (frmDepthZ16 == null)
				nullFrmDepthZ16++;    // <=====  This is always zero. Why?

```

As expected, after `waitForFrames`, there's always a color frame available, so `nullFrmColor` is zero

As the depth stream is recorded at 6fps in contast to the 15fps in the rgb stream, we cannot expect to get a depth frame for each color frame, and so `nullFrmDepthRgb` >> 0

But the question is... Why is `nullFrmDepthZ16` also zero? Shouldn't I expect it to equals `nullFrmDepthRgb`? Should we have the same number of depth frames disregarding its format?

dorodnic, 2020-10-07:
Hi @AMArostegui 
Prior to colorizer filter, inside `frames` you will find frame in RGB format, colour stream and Z16 format, depth stream. 
Applying colorizer filter will replace Z16 depth frame with an RGB depth frame inside resulting `processed`. You can still get the original Z16 frame from `frames` (its doesn't override the frame, just replaces it inside the resulting frameset collection)

AMArostegui, 2020-10-07:
Hi @dorodnic . I think I found part of my answer

The notes for [rs2::pipeline::wait_for_frames](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#ac77cdd3d2f4ce1e85ea30d475743cec8) explains that

 _In case of different frame rates of the streams, the frames set include a matching frame of the slow stream, which may have been included in previous frames set_

So depth frames are repeated and that explains why we have the same number of Z16 and color frames.

But I still don't undestand why, after using colorized filter, trying to get a StreamType.DEPTH frame in StreamFormat.RGB8 format sometimes returns null when a repeated Z16 frame was available to convert.

dorodnic, 2020-10-07:
> But I still don't undestand why, after using colorized filter, trying to get a StreamType.DEPTH frame in StreamFormat.RGB8 format sometimes returns null when a repeated Z16 frame is available for conversion.

That is strange indeed. 

One reason I can think of is related to Java garbage collection. For better performance we reuse up to 16 frames from each type, and once you run out, you will get null or an empty frame. This can happen if you are not explicitly releasing native objects in time and leaving it up to the garbage collector. I'm less familiar with Java, but in all C# examples we either explicitly call Dispose or call `using`.

closed 2020-10-08 by AMArostegui