realsenseai/librealsense #7513: Need detailed exception handling guide to realsense D435 camera

naruto-uzumakki, 2020-10-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  16.04                                     | 
| Platform                                 | NVIDIA Jetson   |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-10-07:
Hi @naruto-uzumakki   Would the error handling documentation for librealsense in the link below meet your needs, please?

https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme

naruto-uzumakki, 2020-10-07:
hi @MartyG-RealSense  i saw the documentation, i need to handle errors like null pointer passed for argument "frame" ,Frame didn't arrive within 15000

MartyG-RealSense, 2020-10-07:
My experience is that RealSense users tend to use a variation of the method for catching errors that is described in that guide to catch most types of exceptions, instead of trying to catch very specific errors.  

**catch (const rs2::error& e)
{
cerr << "Some other error occurred!" << endl;
}**

dorodnic, 2020-10-07:
That is correct 👍

closed 2020-10-08 by naruto-uzumakki