realsenseai/librealsense #7516: Frame didn't arrive within 5000 - With two camera | c# D435

tchenus, 2020-10-07:
Hello everybody,

Since I'm using the D435 camera I'm encountering a problem. I use a configuration with two D435 cameras on the same pc. I use a DLL that I created to create the pipeline and the ProcessingBlock to apply filter. I identifying cameras with their serial number. But on one of the applications I have an error message that appears after a while. And before it crashes I see the image freezes a little bit from time to time, then once it appeared once it freezes even more and the message comes back regularly. The message is the following, the important part it's "Frame didn't arrived within 5000", before it's my message :

[image]

I tried to change the camera, change the usb cable, change the computer, nothing help. What surprises me is that in the application that crashes I removed absolutely all my code and it crashes anyway. The code just displays the stream video now. And I use exactly the same method for the other application and I have no problem.

Someone have any idea why I have my camera crashing?

Thanks a lot

RealSenseSupport, 2020-10-10:
@tchenus Could you please share your code snippet? Also could you please provide below information? Thanks!

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          |

RealSenseSupport, 2020-10-15:
@tchenus Could you please provide more information as previous list so that we can further support on this? Thanks!

RealSenseSupport, 2020-10-21:
@tchenus Could you please update? Thanks!

RealSenseSupport, 2020-10-29:
@tchenus Did you still need support on this? Please notice that we'll close it if no further update from your side for another 7 days. Thanks!

RealSenseSupport, 2020-11-05:
@tchenus Sorry we didn't hear from your for weeks. Will close it at this point. Please feel free to create new one if you still have any other questions or issues. Thanks!

closed 2020-11-05 by RealSenseSupport