realsenseai/librealsense #7517: Logging with C# SDK.

rajrajeswaran, 2020-10-07:
I am currently investigating an issue i am experiencing with my C# application using the following:

Camera: D415
SDK: 2.38.1
Firmware: 5.12.7.100
Windows 10 Driver : 6.11.160.21

I got this recommended configuration from [here](https://dev.intelrealsense.com/docs/firmware-releases?_ga=2.206337299.1672247464.1601518728-297082828.1592498098%23section-recommended-production-configurations).

I am getting the following exception in C#: Failed to get frames Couldn't resolve requests

I would like to know how I can setup logging for the Intel C# SDK to get more details around this error.

When I revert the Windows 10 driver to 6.11.160.14 version (while maintaining other configuration same as above), i am able to get frames just fine in my C# app.

I am hoping logging will reveal more details that i can share here to find the root of the issue.

MartyG-RealSense, 2020-10-07:
Hi @rajrajeswaran   The UWP RealSense driver is not required for using RealSense with Windows.   Its purpose is to interface a RealSense 400 Series camera with products that support Microsoft's **Universal Windows Platform (UWP)**, such as the Microsoft Hololens headset, so most RealSense users do not need to use it.   You can safely remove this driver and RealSense will still work fine on Windows.

rajrajeswaran, 2020-10-07:
> Hi @rajrajeswaran The UWP RealSense driver (**6.11.160.21**) is not required for using RealSense with Windows. Its purpose is to interface a RealSense 400 Series camera with products that support Microsoft's **Universal Windows Platform (UWP)**, such as the Microsoft Hololens headset, so most RealSense users do not need to use it. You can safely remove this driver and RealSense will still work fine on Windows.

@MartyG-RealSense  - Thank you for your quick response.  I do understand that UWP drivers are not used while running Windows Forms applications.  I would like to understand why our forms application is broken with the UWP driver update.  My concern is that our clients might update the windows driver and our application will stop functioning.  I am happy to provide any logs to identify the root cause but i need some help setting up to capture logs at the SDK level.

MartyG-RealSense, 2020-10-07:
I have just reported the issue within Intel.  I have no further update to provide right now.  

The UWP driver is separate from the RealSense SDK and so your clients would have to go looking for this driver themselves and then download and install it in order to be affected.  

Information on setting logging in the librealsense SDK can be found in the link below:

https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-how-do-i-enable-librealsense-logs

rajrajeswaran, 2020-10-09:
> I have just reported the issue within Intel. I have no further update to provide right now.
> 
> The UWP driver is separate from the RealSense SDK and so your clients would have to go looking for this driver themselves and then download and install it in order to be affected.
> 
> Information on setting logging in the SDK can be found in the link below:
> 
> https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-how-do-i-enable-librealsense-logs

rajrajeswaran, 2020-10-09:
@MartyG-RealSense - Can you provide any reference to the issue you logged with intel so that I can track as well?  I am happy to contribute to their investigation if/when necessary.

MartyG-RealSense, 2020-10-09:
I cannot share details of internal processes, though there is no problem with sharing links to the public cases where issues with this UWP driver have been reported.

**Unity wrapper** 
https://github.com/IntelRealSense/librealsense/issues/7425

**Dynamic Calibrator tool**
https://github.com/IntelRealSense/librealsense/issues/7496#issuecomment-703675130

MartyG-RealSense, 2020-10-25:
Case closed due to no further comments received.

closed 2020-10-25 by MartyG-RealSense

rajrajeswaran, 2020-10-30:
@MartyG-RealSense - My apologies for not responding in a timely manner.  I would like to know the resolution of this issue.  Will there be any communication when a resolution is identified?

MartyG-RealSense, 2020-10-30:
There is currently no update regarding the UWP driver that I can provide.   The **Bug Fixes and Enhancements section** of the release notes of SDK releases is usually where bug resolutions are officially announced publicly.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes

MartyG-RealSense, 2022-01-18:
Hi everyone, I have received an update from Intel that UWP is not supported any more.