realsenseai/librealsense #7521: Error updating firmware - (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS

logidelic, 2020-10-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Ubuntu 18  | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.38.1                          | 

### Issue Description
I'm consistently running into this error while trying to update my camera's firmware:

```
$ rs-fw-update -f ./Signed_Image_UVC_5_12_7_100.bin 

updating device: 
Name: Intel RealSense D415, serial number: 932122062056, update serial number: 933623022483, firmware version: 05.11.01.100, USB type: 3.2
 07/10 16:03:14,175 ERROR [548156935328] (types.h:313) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 07/10 16:03:19,106 ERROR [548122046240] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 07/10 16:03:19,106 ERROR [548122046240] (fw-update-device.cpp:128) access failed for 8086:adb uid: 2-3.2-39
 07/10 16:03:19,108 ERROR [548122046240] (context.cpp:431) Exception thrown from user callback handler

failed to locate a device in FW update mode
```

Some notes: Once I get this error, I need to disconnect-reconnect the USB cable or I can't access the camera at all anymore. Aside from this issue, I am able to use the camera just fine...

Any help would be appreciated.

dorodnic, 2020-10-07:
`RS2_USB_STATUS_ACCESS ` is almost always indicate permissions issue, usually caused by not installed udev-rules. The Viewer will prompt you if udev-rules are not properly installed, but you may have missed it. This is also sometimes causing problems when running inside a container, or some other non-standard linux setup.

logidelic, 2020-10-07:
That was it. Thank you so much.

closed 2020-10-07 by logidelic