realsenseai/librealsense #7524: Unity - Get depth at certain X, Y?

indirawrs, 2020-10-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                           | 2.0 | 
| Language                               | Unity | 
| Segment			         | PC | 
----------------------------------------------------------------------------------------------------

Any recommendations for methods/specific functions to access in the Unity package to read and output the depth value at a certain (x,y)? 
I haven't seen a sample that is useful for me, but there are code samples for other languages that would work for me.

I want something like this but with Unity: 
// Create a Pipeline - this serves as a top-level API for streaming and processing frames
```
rs2::pipeline p;
 
// Configure and start the pipeline
p.start();
 
while (true)
{
    // Block program until frames arrive
    rs2::frameset frames = p.wait_for_frames();
 
    // Try to get a frame of a depth image
    rs2::depth_frame depth = frames.get_depth_frame();
 
    // Get the depth frame's dimensions
    float width = depth.get_width();
    float height = depth.get_height();
 
    // Query the distance from the camera to the object in the center of the image
    float dist_to_center = depth.get_distance(width / 2, height / 2);
 
    // Print the distance
    std::cout << "The camera is facing an object " << dist_to_center << " meters away \r";
}
```
Thanks.

MartyG-RealSense, 2020-10-08:
Hi @indirawrs   There is a RealSense tutorial for the Unity wrapper and 400 Series cameras that demonstrates switching between 2D and 3D coordinate systems.  It can map 2D coodinates to a 3D space, or take a 3D point as the input and then output the corresponding 2D coordinates.

https://lightbuzz.com/intel-realsense-coordinate-mapping/

indirawrs, 2020-10-08:
> https://lightbuzz.com/intel-realsense-coordinate-mapping/
> 
> The author explains the code section by section and then posts the complete source code for the tutorial at the bottom of the article, just above the comments section.

Thank you so much for the link! I'll try it out today.

MartyG-RealSense, 2020-10-12:
If you are seeking a way to render data in a more artist-friendly way, the link below contains an example program for linking a RealSense camera to a Unity **Visual Effect Graph**.

https://github.com/keijiro/Rsvfx

Unity Japan produced a Visual Effect Graph presentation about RealSense on YouTube.  The video's subtitle translates to "Combining Visual Effect Graph and RealSense makes it easy to do flashy things and it's really interesting".

https://www.youtube.com/watch?v=uvGX8GPdfTE

indirawrs, 2020-10-15:
I really just need to get X/Y/Z values like the Python examples, but can't find clear tutorials on how to use any examples. Unity is the only thing I've got to work with that I'm familiar with and it doesn't have similar code. It's frustrating.

MartyG-RealSense, 2020-10-15:
I carefully researched again the subject of how to get xyz coordinates in Unity with RealSense but found nothing that rendered coordinate text on-screen.  This may be because the RealSense Unity wrapper applies camera data to a Unity texture to generate the on-screen images.  

I searched the Unity Asset Store for **point cloud** to see how other point cloud viewing solutions in Unity handled coordinates, and all of them only rendered graphics without accompanying text coordinates.

closed 2020-10-30 by MartyG-RealSense