realsenseai/librealsense #7526: RuntimeError: Frame didn't arrive within 5000

JANGSOONMYUN, 2020-10-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1                      | 
| Language                            |  C++/Python                      | 

I have tried to run example codes in the SDK. But image frames cannot be gotten and an error message was shown: 
RuntimeError: Frame didn't arrive within 5000

In the Intel RealSense Viewer, every function works well, but if I run a code on Visual Studio with C++ and Python, it failed.

What should I do for it?
I have tried many solutions in github and stackoverflow.

MartyG-RealSense, 2020-10-08:
Hi @JANGSOONMYUN   It is a positive sign that the camera is detectable by the Python wrapper and the pyrealsense2 module can be found.

Is the **RuntimeError: Frame didn't arrive within 5000** message being generated constantly from the start of the program, or is the program able to function normally for a time before the message occurs?

JANGSOONMYUN, 2020-10-08:
@MartyG-RealSense Thank you for your reply.
I used an example code on Python:
```
import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()
try:
  for i in range(0, 100):
    frames = pipe.wait_for_frames()
    for f in frames:
      print(f.profile)
finally:
    pipe.stop()
```

rs.pipeline() and pipe.start() was able to be run. But I'm not sure those functions ran correctly or not. How could I check results of rs.pipeline() and pipe.start()?

And the error occurred at this line: frames = pipe.wait_for_frames()

MartyG-RealSense, 2020-10-08:
Could you try moving **frames = pipe.wait_for_frames()** outside of the For loop please so that it is one line above it:

import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()
try:
**frames = pipe.wait_for_frames()**
for i in range(0, 100):
    for f in frames:
      print(f.profile)
finally:
    pipe.stop()

JANGSOONMYUN, 2020-10-08:
@MartyG-RealSense 
I tried but same error occurred.

I don't understand why it cannot get a frame...

MartyG-RealSense, 2020-10-08:
i see that you are using the test script from the pip pyrealsense2 site:

https://pypi.org/project/pyrealsense2/

I would recommend instead using the **python-tutorial-1-depth.py**  tutorial script from the Python wrapper's Examples section as your test script and see whether you still get the **Frame didn't arrived within 5000** error.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

JANGSOONMYUN, 2020-10-09:
@MartyG-RealSense 
I think the tutorial (**python-tutorial-1-depth.py**) is just same as the example code I used. 

And the result is same. 
**Frame didn't arrive within 5000**

MartyG-RealSense, 2020-10-09:
Let's try preventing the error from being thrown.   It is possible that data still will not arrive, but we can at least prevent the error message from triggering.   Could you use **try_wait_for_frames()** instead of **wait_for_frames()** please?  The effect of this is that it stops the error from being thrown and just registers a 'false' logic state instead if the stream times out.

JANGSOONMYUN, 2020-10-12:
@MartyG-RealSense 
Sorry for late reply. I couldn't try it during weekends.

I have just tried it but there is no response. It doesn't show any error or exception, but just waiting. I waited for 5 mins, but still no response.

JANGSOONMYUN, 2020-10-12:
@MartyG-RealSense 
Sorry for late reply. I couldn't try it during weekends.

I have just tried it but the message of result is always 'False'.

I actually tried it on my personal laptop, and it worked very well.
I think there is something wrong on the USB driver installation or Registry problem. Could you let me know how to remove the driver and the registry completely?

MartyG-RealSense, 2020-10-12:
Rather than try to remove drivers, it may be better to let Windows' **Troubleshoot** tool do an analysis and see whether it can find a problem, as it will then try to automatically correct it.  Microsoft hide the hardware troubleshoot option on some modern versions of Windows.

To access it now, you have to open the Command Prompt interface and copy and paste the line below into the Command Prompt window.

**msdt.exe -id DeviceDiagnostic**

[image]

[image]

JANGSOONMYUN, 2020-10-14:
@MartyG-RealSense 
I have tried the Troubleshoot tool, but no different result.
Um.. I spent one week for this. Maybe I should use other computer.
Thank you so much for your help.

MartyG-RealSense, 2020-10-14:
You are very welcome.  There is a good chance that using another computer may resolve the problem if that is an option for you, as it is the type of problem that may occur one one machine but not another.

closed 2020-10-15 by JANGSOONMYUN